platformio/platformio-core #2782: Home: Could not initialize project with Rasperry Pi 3 B

Jeronimo016, 2019-07-12:
PIO Core Call Error: "<Exception instance at 0x5b43b48 with str error:\n Traceback (most recent call last):\n  File \"C:\\.platformio\\packages\\contrib-pysite\\twisted\\python\\reflect.py\", line 448, in safe_str\n    return str(o)\nUnicodeEncodeError: 'ascii' codec can't encode character u'\\xf3' in position 60: ordinal not in range(128)\n>"

ivankravets, 2019-07-13:
Do you use the latest PIO Core 4.0?

Jeronimo016, 2019-07-13:
How can i see if I use the latest PIO Core 4.0?

ivankravets, 2019-07-13:
On the PIO Home > Welcome page.

Jeronimo016, 2019-07-13:
I just updated it but now the progarm has a new error:
PIO Core Call Error: "The current working directory C:\\Users\\Jerónimo Torres\\Documents\\PlatformIO\\Projects\\ppppppppppp will be used for the project.\n\r\nThe next files/directories have been created in C:\\Users\\Jerónimo Torres\\Documents\\PlatformIO\\Projects\\ppppppppppp\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: PlatformIO temporary does not support cross-compilation for WiringPi framework. Please use PIO Core directly on Raspberry Pi"

ivankravets, 2019-07-13:
> PlatformIO temporary does not support cross-compilation for WiringPi framework. Please use PIO Core directly on Raspberry Pi

https://github.com/platformio/platform-linux_arm/issues/2

closed 2019-07-13 by ivankravets