platformio/platformio-core #2789: Clion cant find Arduino.h with --ide clion

Ostpopcorn, 2019-07-13:
Hello,
I realy enjoy working i Clion so i followed the tutorial om getting started with Clion + PlatformIO.
Installation was smooth but when when I opened the project I created for the clion ide can't find the Arudino.h file, and it dosent build.
I'll try to make everything readable by a bullet list
* PlatformIO --version -> 4.0.0
* I'm using esp32dev as device `--board esp32dev`
* Arudino.h is not found
* Headers in library folder of "framework-arduinoespressif32" are found by the ide
* Building in CLion fails with this log:
```
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10002.190628 (1.0.2), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Blynk> 0.5.1
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
Building SPIFFS image from 'data' directory to .pio\build\esp32dev\spiffs.bin
error: can't read source directory
*** [.pio\build\esp32dev\spiffs.bin] Error 1
Dependency Graph
|-- <Blynk> 0.5.1
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
 [ERROR] Took 2.23 seconds 
mingw32-make.exe[2]: *** [CMakeFiles\PLATFORMIO_UPLOADFS.dir\build.make:56: CMakeFiles/PLATFORMIO_UPLOADFS] Error 1
mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:72: CMakeFiles/PLATFORMIO_UPLOADFS.dir/all] Error 2
mingw32-make.exe[1]: *** Waiting for unfinished jobs....
Compiling .pio\build\esp32dev\src\main.cpp.o
src\main.cpp:6:21: fatal error: Arudino.h: No such file or directory

*****************************************************************
* Looking for Arudino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Arudino.h"
* Web  > https://platformio.org/lib/search?query=header:Arudino.h
*
*****************************************************************

compilation terminated.
Compiling .pio\build\esp32dev\lib5f2\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32dev\lib5f2\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32dev\lib5f2\WiFi\WiFiGeneric.cpp.o
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
 [ERROR] Took 5.08 seconds 
mingw32-make.exe[2]: *** [CMakeFiles\PLATFORMIO_BUILD.dir\build.make:56: CMakeFiles/PLATFORMIO_BUILD] Error 1
mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:109: CMakeFiles/PLATFORMIO_BUILD.dir/all] Error 2
mingw32-make.exe: *** [Makefile:83: all] Error 2
``` 
No spiffs.bin is where the path points to. 
*Building from CMD fails with 
```
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WiFi> 1.0
|-- <Blynk> 0.5.1
|   |-- <WiFi> 1.0
Compiling .pio\build\esp32dev\src\main.cpp.o
src\main.cpp:6:21: fatal error: Arudino.h: No such file or directory
```
* Compiling projects with VSC works on the same machine (but its a diffirent platformio core..)
* Compiling  projects from a cmd that was not with `--ide clion` completes that step.
* Compling a new project with `--ide clion` and running `platformio run` before it has been opened by Clion also fails.
 
I'll provide more info if needed!

Cheers

ivankravets, 2019-07-15:
It seems like you have broken dev/platform and its packages. Do you have antivirus tools in the system? Try to disable them, remove `~/.platformio/packages` folder and build the project again.

Does it work now?

ivankravets, 2019-07-17:
Please reopen if you still need help

Ostpopcorn, 2019-07-17:
Apologies for not answering sooner. 
I don't run with anything but windows defender at the moment. Opening CLion without perfoming any actions it promted me with 
```Load error: undefined path variables
			FILE_PATH is undefined. Fix it
			Path variables are used to substitute absolute paths in CLion project files and allow project file sharing in version control systems.
			Some of the files describing the current project settings contain unknown path variables and CLion cannot restore those paths.```
After following the step you described and the rebuilding it now produces the same error. Restarting CLion and opening the project agin i get the `FILE_PATH` error described earlier in this post.

ivankravets, 2019-07-19:
Do you use the latest version of CLion?

Ostpopcorn, 2019-07-20:
Yes i do

ivankravets, 2019-07-23:
Could you create a simple project which does not work and attach here in an archive? Thanks!

P.S: I've just tried a simple project and evrything works as expected.

Ostpopcorn, 2019-08-05:
Hi again, I was away for a while but I'm back now. 
I got the update to CLion 2019.2 and platformio was up to date, but it changed nothing and the same error still persists.
[ClionTestMalfunction.zip](https://github.com/platformio/platformio-core/files/3469083/ClionTestMalfunction.zip)

ivankravets, 2019-08-15:
> (but its a diffirent platformio core..)Compiling projects from a cm

Please upgrade to the latest PlatformIO Core 4.0. Does it build a project from CLI?

Ostpopcorn, 2019-08-16:
I ran "pio upgrade" and "pio update" and then made a fresh project in powershell. Still nothing.
I tested just for fun to leave out --ide and it got the same behavior. 
Can it be the different installlations (one installed with pip and the other one from platforio plug-in for VSC) that causes something?

ivankravets, 2019-08-19:
How did you install PlatformIO Core? 

Please provide an output from system Terminal `cmd`
```
echo %PATH%

python --version
pip --version
pip list
```

Ostpopcorn, 2019-08-20:
C:\Users\adrian\platformio\CLion\Test7>echo %PATH%
C:\Program Files\Espressif\ESP-IDF Tools\mconf-idf;C:\Program Files\Espressif\ESP-IDF Tools\tools\bin;C:\Python27\Scripts\;C:\Python27;C:\Program Files\Python37\Scripts\;C:\Program Files\Python37\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\system32;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Users\adrian\.platformio\penv\Scripts;C:\Users\adrian\esp\esp-idf\tools;C:\Program Files\CMake\bin;C:\Program Files\Git\cmd;C:\Users\adrian\AppData\Local\Microsoft\WindowsApps;C:\Users\adrian\AppData\Local\atom\bin;C:\Users\adrian\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\adrian\.dotnet\tools;C:\Users\adrian\AppData\Local\GitHubDesktop\bin;;%USERPROFILE%\AppData\Local\Microsoft\WindowsApps

C:\Users\adrian\platformio\CLion\Test7>
C:\Users\adrian\platformio\CLion\Test7>python --version
Python 2.7.16

C:\Users\adrian\platformio\CLion\Test7>pip --version
pip 19.1.1 from c:\python27\lib\site-packages\pip (python 2.7)

C:\Users\adrian\platformio\CLion\Test7>pip list
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Package          Version
---------------- ---------
-latformio       3.6.7
asn1crypto       0.24.0
bottle           0.12.17
certifi          2019.6.16
cffi             1.12.3
chardet          3.0.4
click            5.1
colorama         0.4.1
cryptography     2.7
enum34           1.1.6
future           0.17.1
idna             2.8
ipaddress        1.0.22
pip              19.1.1
platformio       4.0.0
pycparser        2.19
pyelftools       0.25
pyparsing        2.3.1
pyserial         3.4
requests         2.22.0
semantic-version 2.6.0
setuptools       40.6.2
six              1.12.0
urllib3          1.25.3
WARNING: You are using pip version 19.1.1, however version 19.2.2 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

ivankravets, 2019-08-20:
So, and `pio run -d /path/to/project` does not work?

Ostpopcorn, 2019-08-20:
Nope, think this is the same a before
```
C:\Users\adrian\platformio\CLion\Test7>pio run
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.9.0 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10002.190628 (1.0.2), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WiFi> 1.0
|-- <Blynk> 0.6.1
|   |-- <WiFi> 1.0
Compiling .pio\build\esp32dev\src\main.cpp.o
Compiling .pio\build\esp32dev\lib5f2\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32dev\lib5f2\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32dev\lib5f2\WiFi\WiFiGeneric.cpp.o
src\main.cpp:6:21: fatal error: Arudino.h: No such file or directory

*****************************************************************
* Looking for Arudino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Arudino.h"
* Web  > https://platformio.org/lib/search?query=header:Arudino.h
*
*****************************************************************

compilation terminated.
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
============================================= [ERROR] Took 13.62 seconds =============================================

```

ivankravets, 2019-08-20:
I see. As minimum this is not a problem of CLion. Please remove `user home folder/.platformio/packages` directory and try build again. Looks like antivirus tools block PlatformIO installer

closed 2019-08-22 by ivankravets

Ostpopcorn, 2019-08-23:
So, 
Redoing that with a admin CMD it now build outside in cmd. (With new project)
But in Clion it now says someting about spiffs.bin.
Full log with "BUILD ALL" give the following log
```
C:\Users\adrian\AppData\Local\JetBrains\Toolbox\apps\CLion\ch-0\192.5728.100\bin\cmake\win\bin\cmake.exe --build C:\Users\adrian\platformio\CLion\Test8\cmake-build-debug --target all -- -j 2
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.9.0 > Espressif ESP32 Dev Module
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PLATFORM: Espressif 32 1.9.0 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10002.190628 (1.0.2), tool-esptoolpy 1.20600.0 (2.6.0)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), tool-mkspiffs 2.230.0 (2.30), framework-arduinoespressif32 2.10002.190628 (1.0.2), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Found 29 compatible libraries
Scanning dependencies...
Scanning dependencies...
No dependencies
No dependencies
Building SPIFFS image from 'data' directory to .pio\build\esp32dev\spiffs.bin
error: can't read source directory
*** [.pio\build\esp32dev\spiffs.bin] Error 1
 [FAILED] Took 3.53 seconds 
mingw32-make.exe[2]: *** [CMakeFiles\PLATFORMIO_UPLOADFS.dir\build.make:56: CMakeFiles/PLATFORMIO_UPLOADFS] Error 1
mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:72: CMakeFiles/PLATFORMIO_UPLOADFS.dir/all] Error 2
mingw32-make.exe[1]: *** Waiting for unfinished jobs....
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   3.9% (used 12820 bytes from 327680 bytes)
PROGRAM: [==        ]  15.1% (used 197496 bytes from 1310720 bytes)
 [SUCCESS] Took 5.99 seconds 
mingw32-make.exe: *** [Makefile:83: all] Error 2
[  0%] Built target PLATFORMIO_BUILD
```
main.cpp is
```
#include <Arduino.h>
void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
}
```
Cheers

ivankravets, 2019-08-29:
No need to build ALL, please select only 1 target in CLion.

Ostpopcorn, 2019-08-29:
Oh nice, it works now! Thanks for keeping out with me :)

ivankravets, 2019-08-29:
New PIO Core 4.0.3 is coming today. It has super cool multi-environment project support.