platformio/platformio-core #2791: Home: Libraries: Stats

Darlingtone, 2019-07-14:
PIO Core Call Error: "Error: [API] Could not connect to PlatformIO API Service. Please try later."

Am getting this Error a lot, installing Library is a hit or miss. i have to retry and retry sometimes for hours before i can install a library

ivankravets, 2019-07-15:
> "Error: [API] Could not connect to PlatformIO API Service. Please try later."

Do you have unstable internet connection?

Darlingtone, 2019-07-15:
My internet is fairly stable, that was the first thing i checked given the nature of the error message

ivankravets, 2019-07-16:
Could you run in PlatformIO IDE Terminal:
```
python -c "import requests; print(requests.get('https://api.platformio.org/lib/stats').json())"
```

Try to run this command multiple times. Does it work every time?

closed 2019-07-23 by ivankravets