platformio/platformio-core #2795: Home: Could not load boards list

Copen-08, 2019-07-16:
PIO Core Call Error: "Error: Traceback (most recent call last):\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 88, in main\n    cli(None, None, None)\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 764, in __call__\n    return self.main(*args, **kwargs)\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 717, in main\n    rv = self.invoke(ctx)\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 37, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 956, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 555, in invoke\n    return callback(*args, **kwargs)\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 30, in cli\n    return _print_boards_json(query, installed)\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 80, in _print_boards_json\n    for board in _get_boards(installed):\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 75, in _get_boards\n    return pm.get_installed_boards() if installed else pm.get_all_boards()\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 191, in get_all_boards\n    boards = self.get_installed_boards()\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\platformio\\util.py\", line 70, in wrapper\n    self.cache[key] = (time.time(), func(*args, **kwargs))\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 179, in get_installed_boards\n    p = PlatformFactory.newPlatform(manifest['__pkg_dir'])\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 260, in newPlatform\n    _instance = platform_cls(join(platform_dir, \"platform.json\"))\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 497, in __init__\n    self._manifest = util.load_json(manifest_path)\n  File \"c:\\users\\ap\\.platformio\\penv\\lib\\site-packages\\platformio\\util.py\", line 124, in load_json\n    with open(file_path, \"r\") as f:\nIOError: [Errno 2] No such file or directory: 'C:\\\\Users\\\\AP\\\\.platformio\\\\platforms\\\\ArduinoXInput_AVR\\\\platform.json'\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

Copen-08, 2019-07-16:
I tried fixing this by reinstalling VS Code, and Platform IO, but it just happened again once I reinstalled it all...

ivankravets, 2019-07-16:
> C:\\Users\\AP\\.platformio\\platforms\\ArduinoXInput_AVR\\platform.json

What is it??? Please remove `C:\\Users\\AP\\.platformio\\platforms\\ArduinoXInput_AVR` and restart VSCode.

closed 2019-07-16 by ivankravets