platformio/platformio-core #2799: Home: Could not initialize project

roboter, 2019-07-16:
PIO Core Call Error: "The current working directory C:\\Users\\robby\\Documents\\PlatformIO\\Projects\\STM32_429_Demo_Cube will be used for the project.\n\r\nThe next files/directories have been created in C:\\Users\\robby\\Documents\\PlatformIO\\Projects\\STM32_429_Demo_Cube\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing disco_f429zi (platform: ststm32; board: disco_f429zi; framework: stm32cube)\n\nVerbose mode can be enabled via `-v, --verbose` option\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/disco_f429zi.html\nPLATFORM: ST STM32 5.4.3 > ST 32F429IDISCOVERY\nHARDWARE: STM32F429ZIT6 180MHz, 256KB RAM, 2MB Flash\nDEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)\nPACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-stm32cube 2.0.181130\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/disco_f429zi.html\nPLATFORM: ST STM32 5.4.3 > ST 32F429IDISCOVERY\nHARDWARE: STM32F429ZIT6 180MHz, 256KB RAM, 2MB Flash\nDEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)\nPACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-stm32cube 2.0.181130\nLDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf\nLDF Modes: Finder ~ chain, Compatibility ~ soft\nLDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf\nLDF Modes: Finder ~ chain, Compatibility ~ soft\nFound 27 compatible libraries\nScanning dependencies...\nNo dependencies\nWindowsError: [Error 183] Cannot create a file when that file already exists: 'C:\\\\Users\\\\robby\\\\Documents\\\\PlatformIO\\\\Projects\\\\STM32_429_Demo_Cube\\\\.pio\\\\libdeps\\\\disco_f429zi':\n  File \"C:\\Users\\robby\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\main.py\", line 126:\n    env.SConscript(\"$BUILD_SCRIPT\")\n  File \"C:\\Users\\robby\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 541:\n    return _SConscript(self.fs, *files, **subst_kw)\n  File \"C:\\Users\\robby\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 250:\n    exec _file_ in call_stack[-1].globals\n  File \"C:\\Users\\robby\\.platformio\\platforms\\ststm32\\builder\\main.py\", line 91:\n    target_elf = env.BuildProgram()\n  File \"C:\\Users\\robby\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 224:\n    return self.method(*nargs, **kwargs)\n  File \"C:\\Users\\robby\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 124:\n    _build_project_deps(env)\n  File \"C:\\Users\\robby\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 49:\n    project_lib_builder = env.ConfigureProjectLibBuilder()\n  File \"C:\\Users\\robby\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 224:\n    return self.method(*nargs, **kwargs)\n  File \"C:\\Users\\robby\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\piolib.py\", line 1043:\n    project.install_dependencies()\n  File \"C:\\Users\\robby\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\piolib.py\", line 875:\n    self.env.subst(join(\"$PROJECTLIBDEPS_DIR\", \"$PIOENV\")))\n  File \"c:\\users\\robby\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\lib.py\", line 39:\n    super(LibraryManager, self).__init__(package_dir)\n  File \"c:\\users\\robby\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 621:\n    os.makedirs(self.package_dir)\n  File \"c:\\users\\robby\\.platformio\\penv\\lib\\os.py\", line 157:\n    mkdir(name, mode)\n [ERROR] Took 191.47 seconds"

ivankravets, 2019-07-17:
> [Error 183] Cannot create a file when that file already exists: 

It seems like antivirus tools block PlatformIO library manager. Could you reproduce this issue after VSCode restart?

roboter, 2019-07-17:
I can confirm that there no file prior creating project, also mbed version worked.

ivankravets, 2019-07-17:
Could you reproduce this issue again after IDE restart?

roboter, 2019-07-17:
I will try.

roboter, 2019-07-17:
Project is created without errors but everything is empty, no main.c/cpp file

Platfrom.ini
[env:disco_f429zi]
platform = ststm32
board = disco_f429zi
framework = stm32cube

Expected minimum setup present
at least main.cpp that can be built
and if it cube project .mxproject and .project files needed

ivankravets, 2019-07-17:
We don't create these files. We don't need them. PlatformIO has a standalone builder for STM32Cube project.

You can get example code from https://github.com/platformio/platform-ststm32/tree/develop/examples

closed 2019-07-17 by ivankravets