platformio/platformio-core #2803: Home: Could not initialize project

nkuck, 2019-07-17:
Very new to PlatformIO. Keep getting the following and not sure what to change within the platform. Help, please.

PIO Core Call Error: "The current working directory /Users/noelkuck1/Documents/PlatformIO/Projects/Blinky will be used for the project.\n\nThe next files/directories have been created in /Users/noelkuck1/Documents/PlatformIO/Projects/Blinky\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing uno (platform: atmelavr; board: uno; framework: arduino)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\n  File \"/Users/noelkuck1/.platformio/platforms/atmelavr/builder/main.py\", line 30\n\n    print \"Please unplug/plug device ...\"\n\n                                        ^\n\nSyntaxError: Missing parentheses in call to 'print'. Did you mean print(\"Please unplug/plug device ...\")?\n\n========================== [ERROR] Took 0.73 seconds =========================="

ivankravets, 2019-07-18:
It seems that you use very old Atmel AVR dev/platform. Please open PIO Home > Platforms > Updates.

nkuck, 2019-07-19:
No, everything latest versions available.