platformio/platformio-core #2811: Home: Libraries: Could not install library

cristianro99, 2019-07-20:
PIO Core Call Error: "Library Storage: C:\\Users\\Cristy\\.platformio\\lib\nLibraryManager: Installing id=1648\nDownloading\nDownloading\nUnpacking\nUnpacking\nEnviroDIY_ModularSensors @ 0.23.2 has been successfully installed!\nInstalling dependencies\nLooking for EnviroDIY_DS3231 library in registry\r\nFound: https://platformio.org/lib/show/2079/EnviroDIY_DS3231\nLibraryManager: Installing id=2079 @ =1.3.2\r\nDownloading\nDownloading\nUnpacking\nUnpacking\nEnviroDIY_DS3231 @ 1.3.2 has been successfully installed!\nLooking for EnableInterrupt library in registry\r\nFound: https://platformio.org/lib/show/311/EnableInterrupt\nLibraryManager: Installing id=311 @ =1.0.0\r\nDownloading\nDownloading\nUnpacking\nUnpacking\nEnableInterrupt @ 1.0.0 has been successfully installed!\nLooking for SdFat library in registry\r\nFound: https://platformio.org/lib/show/322/SdFat\nLibraryManager: Installing id=322 @ =1.1.0\r\nDownloading\nDownloading\nUnpacking\nUnpacking\nSdFat @ 1.1.0 has been successfully installed!\nLibraryManager: Installing Adafruit ADS1X15\n\n\nError: Please install Git client from https://git-scm.com/downloads"

ivankravets, 2019-07-23:
Error: Please install Git client from https://git-scm.com/downloads"

closed 2019-07-23 by ivankravets

shenwwyy, 2019-12-04:
i was install git,and restart platformio,but it also can't install paltfrorm

shenwwyy, 2019-12-04:
i was install git in ubuntu,and restart platformio,but it also can't install paltform

thecriticalpoint, 2020-12-06:
What a nightmare

enekochan, 2021-02-25:
In my case `git` would broke after a macOS update is installed. Any `git` command would output this:

```sh
$ git --version
xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at:
/Library/Developer/CommandLineTools/usr/bin/xcrun
```

Reinstalling the "Command Line Tools" fixes it:

```sh
$ xcode-select --install
```

tablatronix, 2022-01-10:
I guess xcode clobbers paths?

shenwwyy, 2022-01-10:
您发来的邮件已收到，谢谢