platformio/platformio-core #2820: Update contributing guidelines

Teo-CD, 2019-07-23:
### Configuration

**Operating system**: Linux Antergos, kernel 5.2.2-arch1-1-ARCH amd64

**PlatformIO Version** (`platformio --version`): 4.0.1b3

### Description of problem

Since https://github.com/platformio/platformio-core/commit/a60c57ac58b4a6902f971bc0f145c89baa153c1c#diff-b91f3d5bd63fcd17221b267e851608e8 the contributing guidelines are not valid anymore as the tox environment "develop" does not exist anymore.

#### Steps to Reproduce

1. Follow contributing guidelines up to `4. Go to the root of project where is located tox.ini and run tox -e develop`

### Actual Results

`ERROR: unknown environment 'develop'`

### Expected Results

Setting up of the development environment.

closed 2019-07-23 by ivankravets

Teo-CD, 2019-07-24:
Hello again. I am sorry I did not try everything in the guidelines at the moment of writing the issue, but there are still some issues :

 - Tox environment `lint` does not exist anymore, step 7 cannot be executed.
 - No mention of the required `git submodule update --init --recursive` to be able to run `tox -e docs`. (To be honest, I did not notice the submodules when I looked at the repo on github and had to search for a bit, but that's on me)

ivankravets, 2019-07-24:
Thanks! updated.

commit ba0ab796 (2019-07-23): Updated contributing guidelines // Resolve #2820
commit 8ea3909a (2019-07-24): Updated contributing guidelines // Resolve #2820
commit 436a6bc5 (2019-07-24): Updated contributing guidelines // Resolve #2820