platformio/platformio-core #2824: Feature request : Better integration of environments in CLion

Teo-CD, 2019-07-24:
Hi. I love platformio and have been using it for two years with CLion without a hitch. Recently I have started using multiple environments but they do not integrate at all in CLion :
 - Cannot run individual environments from the IDE
 - Code completion features are based on the first environment only

Those issues forced me to add a dummy environment, write some custom CMake and have some manual setup after each `pio init`, even though I felt that this could be integrated in platformio.

In the process of solving this issues for my project, I found ways around them which seemed very portable and thought that they might help enhance platformio's integration with CLion.

I forked the repo, tried some things around and I managed to solve the above-mentioned issues. Should I submit a PR ?

nexoscp, 2019-08-14:
Started a CLion Plugin at  [github.com/nexoscp/PlatformIOLion](https://github.com/nexoscp/PlatformIOLion). Help is welcome!

ivankravets, 2019-08-17:
@nexoscp  wow!!! Do you need any help from our side?

ivankravets, 2019-08-18:
@Teo-CD Great work on https://github.com/platformio/platformio-core/compare/develop...Teo-CD:clion-integration?expand=1

I've just refactored a little bit ProjectGenerator (see https://github.com/platformio/platformio-core/commit/1dc15326c9d5d3e894e20d5ed2dfc89d4c9358e1) and added `config` to template generator.

So, you have full access to ProjectConfig via `{{ config }}` of `config` varaibles. For example, `config.envs()` returns avaialble environemnts. See [ProjectConfig API](https://github.com/platformio/platformio-core/blob/develop/platformio/project/config.py).

Could you remove your branch and create a new one based on `config`? We would be happy to merge it. Thanks!

nexoscp, 2019-08-19:
> @nexoscp wow!!! Do you need any help from our side?

Help is always welcome. I'm new to PlatformIO so having  a feature list to catch up withe the other IDEs would be great.
To complete the platform.ini-editor, a specification of sections, keys, type of values and default values would be also great.  Also some sort of schema for the JSON-outputs generated by the platformio cli. This should also help other integration projects.

ivankravets, 2019-08-19:
I think the first what should be implemented is PlatofrmIO Core installer. We wrote it in JS and some part of it is ported to Java by @gholdys https://github.com/platformio/platformio-eclipse-ide/tree/master/src/com/insightmachines/pio/installer

The original source code is located here https://github.com/platformio/platformio-node-helpers/tree/master/src/installer

What does this installer do? It creates a Python virtual environment using different methods and installs PIO Core. So, it means that no need to ask a user to install PIO Core manually. When user starts Eclipse/CLIon, the extension checks PIO Core installation and does everything automatically. They, the extension uses it for PIO Core calls.

--------

# The question

Sorry, I'm not Java developer. Is it possible to have something similar to `platformio-node-helpers` but for Java? `platformio-java-helpers`? So, other people can re-use these helpers for own needs. Also, this package can be used by Eclipse or CLion extension.

Thanks!

nexoscp, 2019-08-19:
> Sorry, I'm not Java developer. Is it possible to have something similar to `platformio-node-helpers` but for Java? `platformio-java-helpers`? So, other people can re-use these helpers for own needs. Also, this package can be used by Eclipse or CLion extension.

And i'm not firm with python, but that should not stop us. Will look into it the next days. For now i have to pack my stuff for [CCCamp](https://events.ccc.de/camp/2019/wiki/Main_Page).  If someone want to meet and chat ping me on [@nexoscp](https://twitter.com/nexoscp)

ivankravets, 2019-08-20:
You don't need to know Python :) We use PlatformIO from IDE via CLI. Reading JavaScript code could help to make this `platformio-java-helpers`.

nexoscp, 2019-08-20:
Had a look at [platformio-eclipse-ide](https://github.com/platformio/platformio-eclipse-ide/tree/master/src/com/insightmachines/pio/installer). Thx  @gholdys !

I see two tasks here: 

- define how to interact with user. Should there be a dialog asking to install PlatformIO or a systenm similar to SDKs in Intellij-Idea or ...  
- extract a eclipse independent jar and publish it to a maven repo

nexoscp, 2019-08-20:
addad a Gradle-Build to publish a JAR: [PR](https://github.com/platformio/platformio-eclipse-ide/pull/1)

gholdys, 2019-08-20:
I'm glad someone will take-over the development of this plugin. @nexoscp, feel free to change the package name. I used my default one, but it probably should be something more PlatformIO oriented.

gholdys, 2019-08-20:
I have some draft plugin code with a view that contains a browser. I've managed to successfully open the home page of a local PlatformIO build server in this browser. I'll commit it by the end of this week.

nexoscp, 2019-08-20:
is there already a established package name for PlatformIO on JVM? If not, is `org.platformio` ok?

gholdys, 2019-08-20:
That would be the most proper package name (at least according to the convention for Java packages)

ivankravets, 2019-08-22:
Thanks @gholdys and @nexoscp!

Should we create a separate repository for `platformio-java-helpers`? So, we can keep here base functionality which later could be reused in Eclipse or CLion extensions. Such as PIO Core installation, project task manager, etc.

nexoscp, 2019-08-22:
Good idea! +1

Teo-CD, 2019-08-23:
Thank you @ivankravets ! I really like the changes you made to ProjectGenerator, I'll create a branch based on `config` as soon as I have the time.

Really nice to see the start of a CLion plugin ! I hope you will succeed with this endeavour, a CLion plugin would be awesome to have! :smiley: @nexoscp

closed 2019-08-29 by ivankravets

ivankravets, 2019-08-29:
@Teo-CD great work! I like the new integration! PIO Core 4.0.3 is coming today 😊

ivankravets, 2019-08-29:
@nexoscp could you contact me@ikravets.com? I would like to discuss what can we do with better integration with CLIon.

Teo-CD, 2019-08-29:
Thank you @ivankravets , I am happy that it worked out nicely !  Really glad to have contributed :D This experience has cemented my wish to contribute more to open source projects - and to platformio if the occasion arises!

Should I update CLion's page on the documentation to document the new integration ?

ivankravets, 2019-08-29:
> Should I update CLion's page on the documentation to document the new integration ?
Yes, it will be great! https://github.com/platformio/platformio-docs/blob/develop/ide/clion.rst

------

Could you upgrade PIO Core to 4.0.3rc via `pio upgrade --dev` and re-test? Do you know how can we fix an issue when environment name contains spaces? For example, `[env:hello world]

Teo-CD, 2019-08-29:
Going to do that right now.

My guess would be to quote env names in a way that CMakes does not leave out the space. I'm going to look into it

Teo-CD, 2019-08-29:
@ivankravets Everything works as expected apart from when the env names as a space in it as you mentioned. 
Changing `{{ env }}` to `"{{ env }}"` in https://github.com/platformio/platformio-core/blob/386883fbe53792c939f79c15d3eb3cb1a8a3b372/platformio/ide/tpls/clion/CMakeListsPrivate.txt.tpl#L50 and https://github.com/platformio/platformio-core/blob/386883fbe53792c939f79c15d3eb3cb1a8a3b372/platformio/ide/tpls/clion/CMakeListsPrivate.txt.tpl#L67 should do the trick

ivankravets, 2019-08-30:
Thanks! It worked!

Teo-CD, 2019-08-30:
@ivankravets Re-tested project generation for CLion from an empty project, it mainly completes but produces an error message as `ide_data` does not exist if there are no leftover environements :
https://github.com/platformio/platformio-core/blob/db6f98336463493bf26deed666c2134dc0aed26c/platformio/ide/tpls/clion/CMakeListsPrivate.txt.tpl#L62-L64

I don't know the cleanest way to handle this but adding an `else` clause with `ide_data = dict()` fixed it. I guess defining `ide_data = dict()` before the `if` would work too.

ivankravets, 2019-08-30:
Thanks, fixed!

Teo-CD, 2019-08-30:
@ivankravets Sorry to be slow to find the bugs, but there is another one: when there are exactly two environments, Python throws the error 
```
File "[...]/platformio/project/helpers.py", line 220, in load_project_ide_data
    if len(envs) == 1 and envs[0] in data:
TypeError: 'set' object does not support indexing
```
when called from https://github.com/platformio/platformio-core/blob/db6f98336463493bf26deed666c2134dc0aed26c/platformio/ide/tpls/clion/CMakeListsPrivate.txt.tpl#L63 .

This one I do not know how to fix.

ivankravets, 2019-08-30:
4.0.3 is out https://github.com/platformio/platformio-core/releases/tag/v4.0.3 and the last bug is fixed in 4.0.4a1. 

If this is a critical issue, we will release 4.0.4 next week.

Teo-CD, 2019-08-30:
Existing projects with 2 environments should be fine, but not benefit from the new integration. You cannot init at all from scratch with 2 environments though. Everything else is behaving as expected (This time, tested it all x) ).

I do not know how many people would be affected, but I suppose not that many ?

commit 1c8666e9 (2019-08-29): Clion integration, resolves #2824 (#2944)

* Better environement integration :
 - Environement can be selected in the build target menu of CLion
 - Platformio target runs on the selected environment
 - Changing environment changes defined preprocessor variables and includes accordingly
 - Added 'All' build profile that runs targets on all environment if there are multiple of them (Original behaviour)

* Calling get_project_dir() only once.

* Fixed include path not being converted to unix style.
Removed duplicate and not normalized  definition
commit 4a6d5e83 (2019-08-29): Added support for multi-environment PlatformIO project for CLion IDE // Resolve #2824 Resolve #2944
commit f966eeb6 (2019-08-30): Fixed an issue with project generator for CLion IDE when 2 environments were used // Resolve #2824