platformio/platformio-core #2825: Feature Request: Ability to set databaseFilename in c_cpp_properties.json

pfeerick, 2019-07-24:
- [x] **Feature Request**.

It would be good if there was some way to set the databaseFilename specified in the auto-generated `c_cpp_properties.json` when using VSCode. Even better if it could be set via a VSCode setting, as well as/or rather than a platformio.ini parameter, as it could then span workspaces, and not need to be constantly (re-)configured. Hence the sometimes small (20MB), sometimes larger (80MB+) database files could be put somewhere with enough storage space. 

Also need to investigate why the following VSCode do not appear to work.

```
"C_Cpp.intelliSenseCachePath": <string>
"C_Cpp.intelliSenseCacheSize": <number>
```

Forum link: https://community.platformio.org/t/vscode-browse-c-cpp-db-in-each-project/7152/8

Also related : platformio/platformio-vscode-ide/issues/787

ivankravets, 2019-08-17:
Please re-test with `pio upgrade --dev`. We removed this configuration from static file. So, you can control C/CPP settings from VSCode.

For example, I tested with this and it works:
```
{
    "C_Cpp.intelliSenseCachePath": "/tmp/.vscode-cpptools",
    "C_Cpp.default.browse.databaseFilename": "/tmp/.vscode-cpptools/.browse.c_cpp.db"
}
```

closed 2019-08-17 by ivankravets

pfeerick, 2019-08-19:
Yup, that seems to do the trick... With `4.0.1-rc1` and vscode-insiders-1.38 it's letting me set the folder. 

```
    "C_Cpp.intelliSenseCachePath": "R:\\vscode-cpptools",
    "C_Cpp.default.browse.databaseFilename": "R:\\vscode-cpptools/.browse.c_cpp.db"
```

But something is writing 

`"databaseFilename": "${workspaceRoot}/.vscode/.browse.c_cpp.db",`

 back if I close and open VSCode, during the "PlatformIO: IntelliSense Index Rebuild" stage. Any idea if that is something in VSCode, or the vscode-extension, or a bug in the insider build?

ivankravets, 2019-08-19:
> But something is writing

http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system

Please remove other PlatformIO Cores and use built-in in VSCode.

pfeerick, 2019-08-23:
tl;dr It's working now... Worked perfectly on Linux, Windows needed... a reboot? 

I'm running dual environments on Windows with PIO (release) on VSCode, and PIO (development) on VSCode Insiders and the following in the VSCode Insiders user `settings.json` to configure the develop environment. I don't know what was going on with the Windows build, as I booted into Linux, and is behaving perfectly, and I've tried again on Windows and it's working perfectly now, both develop and release. I'll chalk it up to Windows not liking me... it must know it's days as my primary OS are numbered!

```
    "platformio-ide.useBuiltinPIOCore": false,
    "platformio-ide.useDevelopmentPIOCore": true,
    "platformio-ide.customPATH": "C:\\platformio\\dev\\Scripts",
```

commit 1a3720cf (2019-08-17): Ability to set "databaseFilename" for VSCode and C/C++ extension // Resolve #2825