platformio/platformio-core #2829: Home: Could not install platform

t1minator, 2019-07-24:
PIO Core Call Error: "PlatformManager: Installing espressif32@ 3.2.2\n\n\nError: Detected unknown package 'Espressif'"

You mention on https://github.com/espressif/esp-idf/releases this is the latest stable release. I need the WIFI module in this release, is there a different code I need to get this version?
Thanks

Tim

t1minator, 2019-07-24:
I apologize, I do have 1.9 installed. This is very confusing trying to understand. 
I built softap with the latest idf, and moved to platformIO. The esp_wifi is not in your 1.9 (3.2.2) release. I am frustrated as I want to use platformIO (cuz it is freaking awesome) but I cannot build my project without this component. Any ideas on how I could push forward on PlatformIO?
Thank you!

ivankravets, 2019-08-17:
See available releases of dev/platform https://github.com/platformio/platform-espressif32/releases

See how to use custom version of dev/platform http://docs.platformio.org/en/latest/projectconf/section_env_general.html#platform

closed 2019-08-17 by ivankravets