platformio/platformio-core #2844: WinError 5 Access is denied error thrown when installing a dependency

mlewand, 2019-07-27:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**: Windows 10 (10.0.17763)

**PlatformIO Version** (`platformio --version`): PlatformIO, version 4.0.0

### Description of problem

I wanted to test how PIO handles dependencies before committing to it.

Whenever I install a dependency from GitHub repo (or full git url) the installation fails with:

#### Steps to Reproduce

1. Create a simple PIO project (I used ESP32 devkit, see `platformio.ini`).
1. Execute `pio lib install mlewand/Transformation` in the directory where `platformio.ini` file is located.

### Actual Results

Installation fails with `PermissionError: [WinError 5] Access is denied: 'X:\\Users\\userName\\Documents\\PlatformIO\\Projects\\pio-test-lib2\\.pio\\libdeps\\esp32doit-devkit-v1\\_tmp_installing-syxlieim-package\\.git\\objects\\06\\3bd79ae35475237188c6fcfed3b97aa076600f'`, **however** the lib gets installed and renamed.

I can also properly use the code in the project. But it seems to break the dep installation chain, e.g. adding `--save` will not add it in `platformio.ini`.

```
Library Storage: X:\Users\userName\Documents\PlatformIO\Projects\pio-test-lib2\.pio\libdeps\esp32doit-devkit-v1
LibraryManager: Installing Transformation
git version 2.21.0.windows.1
Cloning into 'X:\Users\userName\Documents\PlatformIO\Projects\pio-test-lib2\.pio\libdeps\esp32doit-devkit-v1\_tmp_installing-syxlieim-package'...
remote: Enumerating objects: 74, done.
remote: Counting objects: 100% (74/74), done.
remote: Compressing objects: 100% (66/66), done.
remote: Total 74 (delta 37), reused 28 (delta 7), pack-reused 0
Unpacking objects: 100% (74/74), done.
Error: Traceback (most recent call last):
  File "X:\Users\userName\.platformio\penv\lib\shutil.py", line 557, in move
    os.rename(src, real_dst)
PermissionError: [WinError 5] Access is denied: 'X:\\Users\\userName\\Documents\\PlatformIO\\Projects\\pio-test-lib2\\.pio\\libdeps\\esp32doit-devkit-v1\\_tmp_installing-syxlieim-package' -> 'X:\\Users\\userName\\Documents\\PlatformIO\\Projects\\pio-test-lib2\\.pio\\libdeps\\esp32doit-devkit-v1\\Transformation'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "X:\Users\userName\.platformio\penv\lib\site-packages\platformio\__main__.py", line 88, in main
    cli(None, None, None)
  File "X:\Users\userName\.platformio\penv\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "X:\Users\userName\.platformio\penv\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "X:\Users\userName\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 37, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "X:\Users\userName\.platformio\penv\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "X:\Users\userName\.platformio\penv\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "X:\Users\userName\.platformio\penv\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "X:\Users\userName\.platformio\penv\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "X:\Users\userName\.platformio\penv\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "X:\Users\userName\.platformio\penv\lib\site-packages\platformio\commands\lib.py", line 149, in lib_install
    force=force)
  File "X:\Users\userName\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 353, in install
    force=force)
  File "X:\Users\userName\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 724, in install
    track=True)
  File "X:\Users\userName\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 523, in _install_from_url
    return self._install_from_tmp_dir(_tmp_dir, requirements)
  File "X:\Users\userName\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 606, in _install_from_tmp_dir
    shutil.move(tmp_dir, pkg_dir)
  File "X:\Users\userName\.platformio\penv\lib\shutil.py", line 569, in move
    rmtree(src)
  File "X:\Users\userName\.platformio\penv\lib\shutil.py", line 507, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "X:\Users\userName\.platformio\penv\lib\shutil.py", line 386, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "X:\Users\userName\.platformio\penv\lib\shutil.py", line 386, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "X:\Users\userName\.platformio\penv\lib\shutil.py", line 386, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "X:\Users\userName\.platformio\penv\lib\shutil.py", line 391, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "X:\Users\userName\.platformio\penv\lib\shutil.py", line 389, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 5] Access is denied: 'X:\\Users\\userName\\Documents\\PlatformIO\\Projects\\pio-test-lib2\\.pio\\libdeps\\esp32doit-devkit-v1\\_tmp_installing-syxlieim-package\\.git\\objects\\06\\3bd79ae35475237188c6fcfed3b97aa076600f'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

```

### Expected Results

Installation process should not throw an error and installation should continue.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**

```ini
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
```

### Additional info

I checked the FAQ (solution 1, 2, 3) but none worked. Actually solution 1 is irrelevant in this case, as the problem happens for local packages dir.

mlewand, 2019-07-28:
After playing a little bit I was able to get around this issue by using WSL (Windows Subsystem for Linux). I installed PlatformIO CLI there and it works fine for the same project.

Although it was a little problematic as `PlatformIO` package didn't want to work without giving a fight.
* couldn't install it for python3.7 in WSL, there was some `ModuleNotFoundError: No module named 'apt_pkg'` errors. Community recommended `sudo apt-get install python3-apt --reinstall` didn't dothe trick.
* couldn't install it via `pip install -U platformio` - it didn't create a global bin `pio` / `platformio`.

Finally what worked out was the magic [super quick installer](https://docs.platformio.org/en/latest/installation.html#super-quick-mac-linux) (but only with py2) which I should have tried in a first place.

From WSL no errors occur and I'm able to install the entire dependency chain, though I'd love to be able to do this in Windows native(ish) shells 😉

Bascy, 2019-07-28:
I get the same errors, practically on installing every single dependency with `pio lib install`
If I run the command again, it continues with the next dependency.

I'm guessing the folder is active in this or some othe process, maybe the pio install command is trying to remove the temp folder while it is still set as working folder or the VSCode IntelliSense process is already running on that new folder and is keeping it occupied

ivankravets, 2019-08-17:
It looks like antivirus tools block build system.

ivankravets, 2019-08-19:
Please reopen if you still need help

knopserl, 2020-04-26:
the problem still exist and no solution has been provided (at least the problem is reproducable in Windows)

mlewand, 2020-04-26:
Yea, I'm still facing it every now and then. It's not caused by AV, as I have disabled it for a given directory and no result.

My workaround for that was that I just bought win10 pro and used WSL to get around this.

ivankravets, 2020-04-27:
Do you use network drive?

knopserl, 2020-05-08:
@ivankravets at least I do not use any network drive. I have also switched off, virus scanner, changed local disk-drives, changed directories, checked/opend permissions, trioed different pc's, new installations of vs code, python, 3.7, 3.8, , pio etc. etc.

ivankravets, 2020-05-09:
Could someone provide a remote session to debug this issue? If yes, please contact us https://platformio.org/contact

closed 2020-05-20 by valeros

drewsed, 2020-09-11:
Disabling my antivirus software did the trick for me.

homandiy, 2021-01-08:
I delete all *.o file at build/uno/src. 
Build again.
Antivirus shows up until I exclude the task.

dr3adx, 2023-02-24:
fcking hell how can nobody fix this access denied error?

romocitto88, 2023-03-07:
I've a similar error: 
"ERROR: Could not install packages due to an OSError: [WinError 5] Access denied: ‘C:\.platformio\penv\Lib\site-packages\~1arset_normalizer\md.cp39-win_amd64.pyd’
Check the permissions."
I've also write in PIO community forum and they suggest me to remove the folder 'C:.platformio\penv' but it didn't solve the issue.
Any ideas?

ivankravets, 2023-03-07:
Do you use antivirus tools?

romocitto88, 2023-03-08:
Yes, but I disabled the antivirus and checked all directories in "C:" if are availabled for all user.
I solved, for now, coping the ".platformio" folder from another PC where it was already installed and working with old library versions (5.2.0).

JackRouss, 2023-05-04:
I have the same issue. tried a bunch of solution I found and nothing work for me. I have Platformio Core 6.1.6 . Anti-virus off, created a exception on anti-virus for .platformio/ .
When I pio run, can't access the charset_normalizer/ folder.
The thing that I observe, is that each time I run it.
The folder change name and seem to have some sort of corrupted characters:
[image]
[image]
I don't know if the first 2 characters always shift like that.

ivankravets, 2023-05-04:
Could you remove `C:/USers/***/.platformio` folder, and create a `.platformio` folder in the root of C drive? `C:/.platformio`. Restart VSCode. Does it work now?

JackRouss, 2023-05-04:
I now have tried to move the .platformio folder to my c:/ dir. didn't work. And I tried to copy a .platformio folder to another pc that the projects build normally.

danielsalyi, 2024-01-24:
Closing VS code solved it, lol.

I had it open, and it was using PIO. Just close your stuff, worst case restart.

commit 73543530 (2020-05-20): Copy and remove cloned package instead of moving // Resolve #2844, Resolve #3328

On Windows, it’s not possible to move a file which is used by another
process (e.g. Git extension in VSCode)