platformio/platformio-core #2848: --force-remote puts src flies into /inc

lakidd, 2019-07-29:
What kind of issue is this?
- [x ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: MacOS Mojave/RasberryPi Rasbian Linux pio_remote 4.19.57+ #1244 Thu Jul 4 18:42:50 BST 2019 armv6l GNU/Linux

**PlatformIO Version** (`platformio --version`): PlatformIO, version 4.0.0 (both ends)

``

### Description of problem
pio remote run -e d1_mini_pro -t upload -v --force-remote builds on remote RPi but puts source code into /include, not /src which leads to 
```
/home/lakidd/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/d1_mini_pro/libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
/home/lakidd/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/d1_mini_pro/libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
/home/lakidd/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/d1_mini_pro/libFrameworkArduino.a(core_esp8266_main.cpp.o): in function `loop_wrapper()':
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x21): undefined reference to `setup'
/home/lakidd/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x2d): undefined reference to `loop'
collect2: error: ld returned 1 exit status
*** [.pio/build/d1_mini_pro/firmware.elf] Error 1

```
#### Steps to Reproduce

1.pio remote run -e d1_mini_pro -t upload -v --force-remote 
2. check directory on remote machine (see below)
3.
If i create a /src and move the cpp file into it, it builds fine on the RPi with a 
`platformio run -t upload`

### Actual Results
See above, linking fails

### Expected Results
Links should succeed

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini

[env:atmega328pb]
platform = atmelavr
board_build.mcu = atmega328pb
;board = elektor_uno_r4
board = ATmega328PB
framework = arduino
lib_extra_dirs= /Users/lakidd/Documents/Arduino/libraries
upload_protocol= usbasp
upload_flags = -Pusb -u -Uhfuse:w:0xDE:m -Uefuse:w:0xF5:m -Ulfuse:w:0xE2:m
board_build.f_cpu = 8000000L
targets = program

[env:d1_mini_pro]
platform = espressif8266
board = d1_mini_pro
framework = arduino
upload_port = /dev/ttyUSB0
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/Blink
*/

void setup(void);
void loop(void);

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(13, OUTPUT);
  Serial.begin(115200);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  Serial.println("BLINK");
  delay(1000);                       // wait for a second
  digitalWrite(13, LOW);
  Serial.println("BLONK");// turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second

}
```

### Additional info
```
lakidd@pio_remote:~/.platformio/remote/projects/Blink_Blonk-c64fc75078561b9c142d275201c2b07d7228527c $ ls
include  lib  platformio.ini
lakidd@pio_remote:~/.platformio/remote/projects/Blink_Blonk-c64fc75078561b9c142d275201c2b07d7228527c $ ls include/
README  main.cpp
lakidd@pio_remote:~/.platformio/remote/projects/Blink_Blonk-c64fc75078561b9c142d275201c2b07d7228527c $ cat include/main.cpp
#include <Arduino.h>

/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/Blink
*/

void setup(void);
void loop(void);

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(13, OUTPUT);
  Serial.begin(115200);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  Serial.println("BLINK");
  delay(1000);                       // wait for a second
  digitalWrite(13, LOW);
  Serial.println("BLONK");// turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second

}
```

ivankravets, 2019-08-12:
Thanks for the report! The bug is fixed in PIO Plus tool 2.5.3. Please re-run `pio update` on the both sides. Restart PIO  Remote agent.

Does it work now?

closed 2019-08-12 by ivankravets

lakidd, 2019-08-12:
Thanks that fixed it, however I'm still struggling with
https://github.com/platformio/platform-espressif8266/issues/165
which the --force-remote was supposed to fix.
Any thoughts there ?

ivankravets, 2019-08-15:
You can install all platform dependecies on remote target via one time using of `--force-remote`. Then, you can build a project on the host side.

lakidd, 2019-08-15:
I've tried that and it still fails when i tried to build host side with
"Error: Could not find LD script"

ivankravets, 2019-08-15:
Could you reopen a new issue with a simple project and instruction to reproduce it? Thanks!

lakidd, 2019-08-15:
Sure, will do