platformio/platformio-core #2857: Sublime/Windows .sublime-project doesn't escape project_dir properly

victorpraxedes, 2019-08-02:
**Configuration**
Windows 10 Pro
Sublime Text 3, Version 3.2.1, Build 3207

**PlatformIO Version (`platformio --version`):**
PlatformIO, version 4.0.0

**Description of problem**
After `pio init --ide sublimetext --board nodemcuv2` the Sublime Text project file `platformio.sublime-project` has the following under `settings`:
```
    "settings":
    {
         "sublimegdb_workingdir": "c:\maelstrom\IosTuff\unit-testing",
         "sublimegdb_exec_cmd": "",
         "sublimegdb_commandline": "C:\\Users\\victor_praxedes\\AppData\\Local\\Programs\\Python\\Python36-32\\Scripts\\platformio.exe -f -c sublimetext debug --interface=gdb --interpreter=mi -x .pioinit"
    }
```
When i try to open the project file `Menu: Project > Open Project...` and select `platformio.sublime-project` it prompts the error:
`Error trying to parse project: Invalid escape in C:\maelstrom\IosTuff\unit-testing\platformio.sublime-project:107:35`

It can be solved by manually escaping the single backslashes on `sublimegdb_workingdir` by adding another backslash, but that should be escaped when the file project is created.

**Steps to reproduce**
```
mkdir unit-testing
cd unit-testing
platformio init --ide sublimetext --board nodemcuv2
```
On Sublime Text 3, `Menu: Project > Open Project...` and select `platformio.sublime-project`

ivankravets, 2019-08-17:
@valeros could you reproduce it?

ivankravets, 2019-08-19:
Please re-test with `pio upgrade --dev`.

Thanks!

commit a2830dd5 (2019-08-19): Automatically normalize file system paths to UNIX-style for Project Generator // Resolve #2857
closed 2019-08-19 by ivankravets