platformio/platformio-core #2860: TypeError on pio remote --taget

tfeldmann, 2019-08-02:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
Remote: Debian (Beaglebone)
Local: macOS 10.14.6 (18G84)

**PlatformIO Version** (`platformio --version`):
Remote and local: PlatformIO, version 4.0.1b3

### Description of problem
pio remote run --upload fails due to a TypeError. Might be a duplicate of #2223.

#### Steps to Reproduce

1. on remote: pio remote agent start
2. on local: pio remote run --target upload
3. "Agent has been terminated" with the tracelog:

```
2019-08-02 14:01:41 [info] Remote command received: psync
Peer will receive following PB traceback:
Unhandled Error
Traceback (most recent call last):
  File "/home/debian/.platformio/packages/contrib-pysite/twisted/spread/banana.py", line 173, in gotItem
    self.callExpressionReceived(item)
  File "/home/debian/.platformio/packages/contrib-pysite/twisted/spread/banana.py", line 136, in callExpressionReceived
    self.expressionReceived(obj)
  File "/home/debian/.platformio/packages/contrib-pysite/twisted/spread/pb.py", line 621, in expressionReceived
    method(*sexp[1:])
  File "/home/debian/.platformio/packages/contrib-pysite/twisted/spread/pb.py", line 1025, in proto_message
    self._recvMessage(self.localObjectForID, requestID, objectID, message, answerRequired, netArgs, netKw)
--- <exception caught here> ---
  File "/home/debian/.platformio/packages/contrib-pysite/twisted/spread/pb.py", line 1054, in _recvMessage
    netResult = object.remoteMessageReceived(self, message, netArgs, netKw)
  File "/home/debian/.platformio/packages/contrib-pysite/twisted/spread/flavors.py", line 125, in remoteMessageReceived
    state = method(*args, **kw)
  File "<string>", line 41, in remote_acwrite

  File "<string>", line 276, in ac_write

  File "/usr/lib/python3.5/json/__init__.py", line 312, in loads
    s.__class__.__name__))
builtins.TypeError: the JSON object must be str, not 'bytes'
```

tfeldmann, 2019-08-05:
Forgot to mention: I'm uploading to a Arduino Due (native USB) from the beaglebone.

rekisum, 2019-08-28:
Got the same problem on macOS and raspbian.
Any remote command leads to:
**a bytes-like object is required, not 'str'**
Both sides updated to **PlatformIO, version 4.0.2.**
Both sides seem to use python 3.
Don't know what to do.

rekisum, 2019-08-29:
Seems the problem is elsewhere .
On the remote side I get:
```
  File "/home/pi/.platformio/packages/contrib-pysite/twisted/spread/flavors.py", line 125, in remoteMessageReceived
    state = method(*args, **kw)
  File "<string>", line 41, in remote_acwrite
    
  File "<string>", line 276, in ac_write
    
  File "/usr/lib/python3.5/json/__init__.py", line 312, in loads
    s.__class__.__name__))
builtins.TypeError: the JSON object must be str, not 'bytes'
```

ivankravets, 2019-09-02:
You use different Python interpreters for Agent and Client. You can check it with `python --version`.

So, you can use Python 2 or Python 3 but you can't mix them.

closed 2019-09-02 by ivankravets

rekisum, 2019-09-03:
Hi, what makes you so sure?
I installed platformio with pip3.
So I hoped it uses python3 by default.
How can I make sure platformio starts with python3?

ivankravets, 2019-09-03:
> File "/home/pi/.

Raspberry Pi uses Python 3, we see that from log above. The client machine uses Python 2.

rekisum, 2019-09-03:
On my raspi:
$ python --version
Python 2.7.9
I got python 2 and 3 on both machines (raspi and mac).
My question is how to let use platformio with the right version.

ivankravets, 2019-09-03:
Try this on Rpi
```
pip3 uninstall platformio
python -m pip install platformio
```

xilart, 2019-12-27:
Hi @ivankravets , unfortunately the problem persists with python3.7 (Win10-PC) on client and python3.5 (RPi). It's interesting that when I had mixed versions then "remote monitor" and "remote device" commands in PIO failed. Now "remote monitor" and "remote device" work fine, but once I try "remote upload" - it fails with two different error types:
1) Client:
`Uploading firmware remotely
Agent has been terminated
The terminal process terminated with exit code: 1`

...and next attempt:
2) Client:
`Uploading firmware remotely
b'dictionary changed size during iteration'
The terminal process terminated with exit code: 1`

Server:
`2019-12-27 16:28:47 [info] Remote command received: psync
2019-12-27 16:28:47 [info] Terminate previous Project Sync process
Peer will receive following PB traceback:
Unhandled Error
Traceback (most recent call last):
  File "/root/.platformio/packages/contrib-pysite/twisted/spread/banana.py", line 173, in gotItem
    self.callExpressionReceived(item)
  File "/root/.platformio/packages/contrib-pysite/twisted/spread/banana.py", line 136, in callExpressionReceived
    self.expressionReceived(obj)
  File "/root/.platformio/packages/contrib-pysite/twisted/spread/pb.py", line 621, in expressionReceived
    method(*sexp[1:])
  File "/root/.platformio/packages/contrib-pysite/twisted/spread/pb.py", line 1025, in proto_message
    self._recvMessage(self.localObjectForID, requestID, objectID, message, answerRequired, netArgs, netKw)
--- <exception caught here> ---
  File "/root/.platformio/packages/contrib-pysite/twisted/spread/pb.py", line 1054, in _recvMessage
    netResult = object.remoteMessageReceived(self, message, netArgs, netKw)
  File "/root/.platformio/packages/contrib-pysite/twisted/spread/flavors.py", line 125, in remoteMessageReceived
    state = method(*args, **kw)
  File "<string>", line 60, in remote_cmd
    
  File "<string>", line 88, in _process_cmd_psync
    
builtins.RuntimeError: dictionary changed size during iteration
`

How can I fix it?

ivankravets, 2019-12-27:
@xilart This is a bug. Please open a new issue. Thanks!