platformio/platformio-core #2867: lib_compat_mode=off is ineffective

sarfata, 2019-08-04:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: macOS 10.14.5

**PlatformIO Version** (`platformio --version`): 4.0.0

### Description of problem

The lib_compat_mode=off option is not respected. Incompatible libraries are still rejected.

#### Steps to Reproduce

```
$ platformio init
$ cat > platformio.ini
[global]
lib_compat_mode = off

[env:kbox32]
platform = https://github.com/platformio/platform-espressif32.git
framework = espidf
board = esp32dev
build_flags = -DKBOX32_SSID=\"YOURWIFI\" -DKBOX32_PASSWORD=\"YOURWIFIPASSWORD\" -fexceptions
lib_compat_mode = off

cat > src/main.cpp
#include <AsyncTCP.h>

int main()
{
  return 0
}

$ platformio run -vv
```

### Actual Results

```
Processing project (platform: espressif32; lib_deps: AsyncTCP; lib_compat_mode: off; board: esp32dev; framework: espidf)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.9.0 #2c1b430 (git+https://github.com/platformio/platform-espressif32.git) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-espidf 3.30202.190627 (3.2.2), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ off
Framework incompatible library /Users/thomas/Desktop/testcompat/.pio/libdeps/project/AsyncTCP_ID1826
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Found 0 compatible libraries
```

### Expected Results

The library should be included in the compilation.

This is what the [documentation](https://docs.platformio.org/en/latest/librarymanager/ldf.html#ldf-compat-mode) says:

> * off: | Does not check for compatibility (is not recommended)
> * soft: [DEFAULT] Checks for the compatibility with framework from build environment
> * strict: Checks for the compatibility with framework and platform from build environment.
> 
> This mode can be changed using lib_compat_mode option in “platformio.ini” (Project Configuration File). Default value is set to soft.

pfeerick, 2019-08-05:
It sort of is and isn't working... the library has `libCompatMode` set to 2, so `strict`, which is overriding what is set in `platformio.ini` ...

[image]

It seems like the checks that happen here get overridden by the library settings if present...

https://github.com/platformio/platformio-core/blob/d9b6842b6a7bca98c4a804c009fd905ff740047e/platformio/builder/tools/piolib.py#L933-L951

... which isn't the behavior I would have expected...

sarfata, 2019-08-07:
@pfeerick thanks for looking into this!

ivankravets, 2019-08-19:
Thanks for the report! Please re-test with `pio upgrade --dev`.

closed 2019-08-19 by ivankravets

sarfata, 2019-08-21:
Looks good to me! Thanks for fixing this!

mcspr, 2019-08-23:
https://github.com/platformio/platformio-core/commit/42ee6fe96e5efb45023780f313f91535108b9d73 seems to break ESPAsyncWebServer builds:
https://travis-ci.org/xoseperez/espurna/jobs/575774509#L515
`pio run -v` shows AsyncTCP as a dependency, which it tries to build instead of ignoring.

I found a typo https://github.com/platformio/platformio-core/commit/fbf9146b
But I think that needs a proper test attached too... or a different method of finding out the default, bc `mode == "strict"` check will fail to invalidate the dependency
```python
> print(validate_compat_mode('deep'))
soft
> print(validate_ldf_mode('strict'))
chain
```

ivankravets, 2019-08-23:
@mcspr thanks again for your reports! They are so valuable. See related issue https://github.com/platformio/platformio-core/issues/2940

It's my typo :-( Need to better cover with tests LDF or add to our  CI a few popular projects, such as ESPurna.

The PIO Core 4.0.2 is going to be released in the next 10-15 minutes.

commit 42ee6fe9 (2019-08-19): Fixed an issue when `library.json` had priority over project configuration for LDF // Resolve #2867