platformio/platformio-core #2871: Home: Could not load boards list

heechul, 2019-08-05:
PIO Core Call Error: "Error: Traceback (most recent call last):\n  File \"c:\\users\\heechul\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 88, in main\n    cli(None, None, None)\n  File \"c:\\users\\heechul\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 764, in __call__\n    return self.main(*args, **kwargs)\n  File \"c:\\users\\heechul\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 717, in main\n    rv = self.invoke(ctx)\n  File \"c:\\users\\heechul\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 37, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"c:\\users\\heechul\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\heechul\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 956, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"c:\\users\\heechul\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 555, in invoke\n    return callback(*args, **kwargs)\n  File \"c:\\users\\heechul\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 30, in cli\n    return _print_boards_json(query, installed)\n  File \"c:\\users\\heechul\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 80, in _print_boards_json\n    for board in _get_boards(installed):\n  File \"c:\\users\\heechul\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 75, in _get_boards\n    return pm.get_installed_boards() if installed else pm.get_all_boards()\n  File \"c:\\users\\heechul\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 191, in get_all_boards\n    boards = self.get_installed_boards()\n  File \"c:\\users\\heechul\\.platformio\\penv\\lib\\site-packages\\platformio\\util.py\", line 70, in wrapper\n    self.cache[key] = (time.time(), func(*args, **kwargs))\n  File \"c:\\users\\heechul\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 180, in get_installed_boards\n    for config in p.get_boards().values():\n  File \"C:\\Users\\heechul\\.platformio\\platforms\\sifive\\platform.py\", line 31, in get_boards\n    result[key] = self._add_default_debug_tools(result[key])\n  File \"C:\\Users\\heechul\\.platformio\\platforms\\sifive\\platform.py\", line 95, in _add_default_debug_tools\n    sdk_dir, \"bsp\", \"sifive-%s\" % board.id, \"openocd.cfg\")\n  File \"c:\\users\\heechul\\.platformio\\penv\\lib\\ntpath.py\", line 65, in join\n    result_drive, result_path = splitdrive(path)\n  File \"c:\\users\\heechul\\.platformio\\penv\\lib\\ntpath.py\", line 115, in splitdrive\n    if len(p) > 1:\nTypeError: object of type 'NoneType' has no len()\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

valeros, 2019-08-05:
Hi @heechul! Can you please describe when this happens? I see it has something to do with SiFive platform, but I can't reproduce it. Is there any custom board added?

heechul, 2019-08-05:
No custom board added. 
After reinstalling the sifive framework, however, the problem went away.

valeros, 2019-08-05:
Thanks for the info, feel free to reopen this issue if the problem occurs again.

closed 2019-08-05 by valeros