platformio/platformio-core #2872: Use XDG_CONFIG_HOME for .platformio directory

microbug, 2019-08-05:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: macOS 10.14

**PlatformIO Version** (`platformio --version`): 4.0.1b3

### Description of problem
The `~/.platformio` directory contains important config data including the user's custom boards. It's currently one of many hidden directories in my home folder. Conventionally these directories go in `$XDG_CONFIG_HOME`, which is usually `~/.local/share`. This makes things neater for the user.

On Windows and macOS the variables `XDG_CONFIG_HOME` usually aren't defined, so you can use `C:\Users\x\AppData\Roaming` and `~/.local/share` respectively (i.e., `~/.local/share` does exist on macOS but isn't usually defined as `XDG_CONFIG_HOME`).

#### Steps to Reproduce

1. Run `pio init`
2. Observe new `.platformio` directory in `$HOME`
3.

peterhoeg, 2019-08-08:
```$XDG_CONFIG_HOME``` is usually ```~/.config``` while ```$XDG_DATA_HOME``` is ```~/.local/share```.

ivankravets, 2019-08-17:
Yes, we keep the same location through all systems. You can override it with http://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_CORE_DIR

lhanson, 2019-10-21:
I think that allowing the user to override PIO's defaults is besides the point; that's a whole lot of directories which a user is forced to remap. The goal of the [XDG Base Directory Specification](https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html) is to free the user from the burden of having to manually manage each piece of software as a unique snowflake directory tree dumped somewhere random in their $HOME directory.

For example, what subdirectories of `~/.platformio` are configuration which needs to be backed up and can be synced between development machines? What directories possibly contain credentials which should never be backed up? Which are runtime data which is necessary to have on hand? Which are cache directories and can safely be discarded if disk space is low? The XDG spec is a way to avoid the user having to figure out the answers to those questions for each piece of software we install. If PIO respected the preferences I've already expressed (and shouldn't have to say differently for every program I install), I'd know that `~/.config/platformio/*` is configuration that I should back up, `~/.cache/platformio` is safe to delete and can be recreated as needed, etc.

Please reconsider opening this issue. Complying with the XDG spec brings a lot of benefits and loses nothing. It's trivial to remain backwards compatible to support existing installations.

ivankravets, 2022-03-31:
See a temporary solution located in the top comment.