platformio/platformio-core #2876: OTA fails after upgrade to 4.0.x

mmichon, 2019-08-06:
### Configuration

**Operating system**:

`MacOS 10.14.6
Darwin Kernel Version 18.7.0: Thu Jun 20 18:42:21 PDT 2019; root:xnu-4903.270.47~4/RELEASE_X86_64`

**PlatformIO Version** (`platformio --version`):
`PlatformIO, version 4.0.0`

### Description of problem
OTA uploads to my ESP32 were working before upgrading to PIO 4.0.x. They no longer work.

#### Steps to Reproduce

1. Issue a `pio run -t upload` to a ESP32 at `esp-entrance.local` (mine is a WROOM equivalent)

### Actual Results
```
mmichon@Mikes-MBAir-2015~/s/e/esp-entrance> pio run -t upload
DEPRECATED! A legacy library storage `/Users/mmichon/src/esp-house/esp-entrance/.piolibdeps` has been found in a project.
Please declare project dependencies in `platformio.ini` file using `lib_deps` option and remove `/Users/mmichon/src/esp-house/esp-entrance/.piolibdeps` folder.
More details -> http://docs.platformio.org/page/projectconf/section_env_library.html#lib-deps
Processing esp32dev (framework: arduino; platform: espressif32; board: esp32dev)
------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.9.0 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), tool-mkspiffs 2.230.0 (2.30), framework-arduinoespressif32 2.10002.190628 (1.0.2), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 60 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <NTPClient> 3.1.0 #1efa610
|-- <FauxmoESP> 2.4.0
|   |-- <AsyncTCP> 1.1.0
|   |-- <WiFi> 1.0
|-- <Time> 1.5 #98c6ec9
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <WiFi> 1.0
|-- <Blynk> 0.6.1
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <WiFi> 1.0
Warning! We have just detected `upload_port` as IP address or host name of ESP device. `upload_protocol` is switched to `espota`.
Please specify `upload_protocol = espota` in `platformio.ini` project configuration file.
Generating partitions .pio/build/esp32dev/partitions.bin
Retrieving maximum program size .pio/build/esp32dev/firmware.elf
Building .pio/build/esp32dev/firmware.bin
Checking size .pio/build/esp32dev/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]  13.4% (used 43912 bytes from 327680 bytes)
PROGRAM: [=======   ]  71.9% (used 941790 bytes from 1310720 bytes)
esptool.py v2.6
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Uploading .pio/build/esp32dev/firmware.bin
20:24:25 [DEBUG]: Options: {'timeout': 10, 'esp_ip': 'esp-entrance.local', 'host_port': 15545, 'image': '.pio/build/esp32dev/firmware.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
20:24:25 [INFO]: Starting on 0.0.0.0:15545
20:24:25 [INFO]: Upload size: 941904
Sending invitation to esp-entrance.local ..........
20:26:05 [ERROR]: No response from the ESP
*** [upload] Error 1
=============================================================== [ERROR] Took 104.90 seconds ===============================================================
```

### Expected Results
Upload successful is expected.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

upload_port = esp-entrance.local
monitor_speed = 115200
lib_deps =
    https://github.com/taranais/NTPClient
    ESPAsyncTCP
    FauxmoESP@2.4.0
    https://github.com/PaulStoffregen/Time
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#include <ArduinoOTA.h>
const char *hostname = "esp-entrance";

void setup(){
    ArduinoOTA.setHostname(hostname);

    ArduinoOTA.begin();
}

void loop(){
    ArduinoOTA.handle();
}
```

### Additional info
This was working with the last 3.x PIO version.

pfeerick, 2019-08-11:
Are you sure it's PIO v4 related? 

Firstly, the warning suggests it is related to the `Switch to "espota" script from Arduino framework` documented in the release notes for the 1.0.9 platformio-espressif32 package. 
```
Warning! We have just detected `upload_port` as IP address or host name of ESP device. `upload_protocol` is switched to `espota`.
Please specify `upload_protocol = espota` in `platformio.ini` project configuration file.
```

Secondly, the port is wrong... it's trying to use 8266 on a ESP32... the correct port is 3232 for  ArduinoOTA with an ESP32 (issue stems from some incorrect documentation from the Arduino ESP32 core that has been fixed and will be corrected in the next release). Update your `platformio.ini` to 

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
upload_protocol = espota
upload_port = esp-entrance.local
upload_flags =
   --port=3232
monitor_speed = 115200
lib_deps =
    https://github.com/taranais/NTPClient
    ESPAsyncTCP
    FauxmoESP@2.4.0
    https://github.com/PaulStoffregen/Time
```
and you should be back in business. Otherwise, you could revert to the prior version of the core by specifying `platform = espressif32@1.8.0`.

mmichon, 2019-08-14:
That did it. I never specified non-default `upload_flags` before so that must've changed with the `platformio-espressif32` changes. They must've been updated on my system at the same time as PIO4, so it was a red herring. Thanks for your help!

closed 2019-08-14 by mmichon