platformio/platformio-core #2881: Home: Could not initialize project

DoniyevShuhrat, 2019-08-08:
Server error: {"args":[2,""],"message":"[Error 2] : u'C:\\\\Users\\\\User\\\\Documents\\\\PlatformIO'","type":"WindowsError"}

DoniyevShuhrat, 2019-08-08:
sorry for wait

pfeerick, 2019-08-11:
I see you have four separate issues open? #2882, #2883, #2884 and this one. Please refrain from posting multiple issues unless there are different problems you're trying to resolve. Feel free to close those other issues. 

Also, a little more detail would be helpful... When did this error come up? What is it preventing you from doing? What troubleshooting steps have you tried? Have you at least looked at the [documentation](https://docs.platformio.org/en/latest/) and [FAQs](https://docs.platformio.org/en/latest/faq.html)?

ivankravets, 2019-08-17:
https://github.com/platformio/platformio-core/issues/2321

closed 2019-08-17 by ivankravets