platformio/platformio-core #2890: Disk character name scrambling

morestart, 2019-08-11:
[image]
OS: win10 1903
Disk character name scrambling.(my disk character is not english)

ivankravets, 2019-10-24:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
python --version
pip --version
pip list
```

morestart, 2019-10-25:
## echo %PATH%

echo %PATH%

## python --version
python --version
## pip --version
pip 19.2.3

## pip list
```
Package              Version
-------------------- ---------------
absl-py              0.7.1
aiohttp              3.5.4
altgraph             0.16.1
appdirs              1.4.3
arrow                0.14.4
asn1crypto           0.24.0
astral               1.10.1
astroid              2.2.5
async-timeout        3.0.1
attrs                19.1.0
backcall             0.1.0
bcrypt               3.1.7
binaryornot          0.4.4
black                19.3b0
bleach               3.1.0
certifi              2019.6.16
cffi                 1.12.3
chardet              3.0.4
Click                7.0
colorama             0.4.1
cookiecutter         1.6.0
cryptography         2.7
decorator            4.4.0
defusedxml           0.6.0
Django               2.2.3
django-filter        2.2.0
djangorestframework  3.10.2
entrypoints          0.3
future               0.17.1
grpcio               1.22.0
homeassistant        0.99.0
idna                 2.8
imageio              2.5.0
importlib-metadata   0.19
ipykernel            5.1.2
ipython              7.7.0
ipython-genutils     0.2.0
isort                4.3.21
jedi                 0.15.1
Jinja2               2.10.1
jinja2-time          0.2.0
json5                0.8.5
jsonschema           3.0.2
jupyter-client       5.3.1
jupyter-core         4.5.0
jupyterlab           1.0.5
jupyterlab-server    1.0.0
lazy-object-proxy    1.4.1
Markdown             3.1.1
MarkupSafe           1.1.1
mccabe               0.6.1
mistune              0.8.4
multidict            4.5.2
nbconvert            5.6.0
nbformat             4.4.0
notebook             6.0.0
numpy                1.16.4
paho-mqtt            1.4.0
pandas               0.25.0
pandocfilters        1.4.2
parso                0.5.1
pefile               2019.4.18
pickleshare          0.7.5
Pillow               6.1.0
pip                  19.2.3
pipenv               2018.11.26
poyo                 0.5.0
prometheus-client    0.7.1
prompt-toolkit       2.0.9
protobuf             3.9.1
pycparser            2.19
Pygments             2.4.2
PyInstaller          3.5
PyJWT                1.7.1
pylint               2.3.1
pyrsistent           0.15.4
python-dateutil      2.8.0
python-slugify       3.0.3
pytorch-lightning    0.4.4
pytz                 2019.2
pywin32-ctypes       0.2.0
pywinpty             0.5.5
PyYAML               5.1.2
pyzmq                18.1.0
requests             2.22.0
ruamel.yaml          0.15.100
Send2Trash           1.5.0
setuptools           41.0.1
six                  1.12.0
sqlparse             0.3.0
tb-nightly           1.15.0a20190708
terminado            0.8.2
test-tube            0.6.9
testpath             0.4.2
text-unidecode       1.2
toml                 0.10.0
torch                1.2.0+cpu
torchvision          0.4.0+cpu
tornado              6.0.3
tqdm                 4.33.0
traitlets            4.3.2
typed-ast            1.4.0
urllib3              1.25.3
virtualenv           16.7.2
virtualenv-clone     0.5.3
voluptuous           0.11.7
voluptuous-serialize 2.2.0
wcwidth              0.1.7
webencodings         0.5.1
Werkzeug             0.15.5
wheel                0.33.4
whichcraft           0.6.0
wrapt                1.11.2
yarl                 1.3.0
zipp                 0.5.2
```

ivankravets, 2019-10-26:
Thanks. But what is your Python version? Do you use Python 3?

morestart, 2019-10-28:
> Thanks. But what is your Python version? Do you use Python 3?

yes i use Python3.7

ivankravets, 2019-10-28:
What is a type of these disks? NFS?

morestart, 2019-10-28:
yes my disk type is ntfs

ivankravets, 2019-10-28:
Please open PlatformIO IDE Terminal (special icon on the bottom toolbar) and type
```
python -c "from platformio import util; print(util.get_logical_devices())"
```

Do you see the correct output?

morestart, 2019-10-28:
i use my frnd win10 computer and i get the result is [{'path': 'C:\\',
'name': 'OS'}, {'path': 'D:\\', 'name': 'ÐÂ¼Ó¾í'}]

ivankravets, 2019-10-28:
I'm not sure that we can fix this. Could you help to debug it? See commands which we run to get disks:
- https://github.com/platformio/platformio-core/blob/develop/platformio/util.py#L165

ivankravets, 2019-10-28:
```
wmic logicaldisk  get name,VolumeName
```

or 

```
fsutil  fsinfo  drives
```

What do you see if tun these commands?

morestart, 2019-10-28:
ok, i can try to fix it, but i have to use my company computer tomorrow, i don't have a win computer...
if i fix it, i will pull request to this project.

ivankravets, 2019-10-28:
Just try to run commands above. What do they return in the console?

morestart, 2019-10-28:
> Just try to run commands above. What do they return in the console?

Name  VolumeName
C:    OS
D:    新加卷

驱动器: C:\ D:\

morestart, 2019-10-29:
> Just try to run commands above. What do they return in the console?

I find the problem, in the proc.py, line 128 , the default code is `result[k] = result[k].decode(sys.getdefaultencoding())` 

but the `sys.getdefaultencoding()` return is utf-8, we just change the utf-8 to GB2312, we can get the correct data..

morestart, 2019-10-29:
> Just try to run commands above. What do they return in the console?
[image]
yes, it worked

ivankravets, 2019-10-29:
Please open PlatformIO IDE Terminal and type 
```
pio upgrade --dev
```

Does it work now? See fix here https://github.com/platformio/platformio-core/commit/0222c56c4da19402c7931882d62d9156f4d4af38

morestart, 2019-10-30:
> Please open PlatformIO IDE Terminal and type
> 
> ```
> pio upgrade --dev
> ```
> 
> Does it work now? See fix here [0222c56](https://github.com/platformio/platformio-core/commit/0222c56c4da19402c7931882d62d9156f4d4af38)

no, this code also returns UTF-8....

morestart, 2019-10-30:
`compact.py`

```python
def get_filesystem_encoding():
    return sys.getfilesystemencoding() or sys.getdefaultencoding()
```

changed to

```python
def get_filesystem_encoding():
    code = subprocess.run('chcp', shell=True, stdout=subprocess.PIPE)
    if (re.findall(r"\d+\.?\d*",str(code.stdout))[-1]) == '936':
        return 'gb2312'
    else:
        return sys.getfilesystemencoding() or sys.getdefaultencoding()
```

it worked

ivankravets, 2019-10-30:
Please re-test with `pio upgrade --dev`.

closed 2019-10-30 by ivankravets

morestart, 2019-10-31:
> Please re-test with `pio upgrade --dev`.

you are right, thank you for your help, it is a excellent work :smile:

commit 0222c56c (2019-10-29): Use file system encoding when decoding subprocess output // Resolve #2890
commit 6d69c25a (2019-10-30): Use locale encoding to decode subprocess output // Resolve #2890