platformio/platformio-core #2904: Feature Request: Inject dependencies into libraries

nomis, 2019-08-15:
- [x] **Feature Request**

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`): 4.0.0

### Description of problem

The `atmelavr` platform with the `arduino` framework does not include the C++ STL.

If a project wants to use C++ STL features, it can depend on the `ArduinoSTL` library.

If a project wants to use a library `X` that requires the C++ STL, then it needs to inject the `ArduinoSTL` library as a dependency of library `X`.

#### Steps to Reproduce

1. Specify both `ArduinoSTL` and library `X` in `lib_deps`
2. Compile the project

### Actual Results

The library `X` is compiled without `ArduinoSTL` as a dependency, which fails because none of the C++ STL header files are in the include path.

### Expected Results

There should be a configuration option to override the dependencies of a library.

### Additional info

The specific choice of C++ STL library should not be made by the maintainers of the `X` library, so it is not sufficient to allow libraries to specify different dependencies per platform.

ivankravets, 2019-08-15:
PlatformIO should handle that automatically. Does library `X` includes any header from `ArduinoSTL`? Could you provide a simple project to reproduce this issue? You can attach project in ZIP archive here.

nomis, 2019-08-15:
[example.zip](https://github.com/platformio/platformio-core/files/3506710/example.zip)

This stops working if I put it **inside** `uuid-common` and make `lib/uuid-common` a symlink to the root of the repository.

I'm trying to verify that the examples that are part of the library work, so I must reference the code in the current repository.

nomis, 2019-08-15:
It works ok if I create `lib/uuid-common/` and then symlink `library.json` and `src` back up to the root, so it must be getting confused by one of the libraries also containing the project.

closed 2019-08-15 by nomis