platformio/platformio-core #2908: Home: Could not initialize project

yuankunli, 2019-08-16:
PIO Core Call Error: "The current working directory D:\\pio-test\\lora-3 will be used for the project.\n\r\nThe next files/directories have been created in D:\\pio-test\\lora-3\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Could not install 'toolchain-gccarmnoneeabi' with version requirements '>=1.60301.0,<1.80000.0' for your system 'windows_amd64'.\n\nPlease try this solution -> http://bit.ly/faq-package-manager"

ivankravets, 2019-08-17:
Please try this solution -> http://bit.ly/faq-package-manager

It seems like antivirus or firewall block PlatformIO package manager.

closed 2019-08-17 by ivankravets

yuankunli, 2019-09-05:
> Please try this solution -> http://bit.ly/faq-package-manager
> 
> It seems like antivirus or firewall block PlatformIO package manager.

Thank you very much!