platformio/platformio-core #2932: Remote build fails with "Could find LD script" --force-remote does not fix

lakidd, 2019-08-21:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
MacOS 10.14.5/ Rasbian Linux pio_remote 4.19.57+ #1244 Thu Jul 4 18:42:50 BST 2019 armv6l GNU/Linux
**PlatformIO Version** (`platformio --version`):
PlatformIO, version 4.0.0 both ends
### Description of problem
When trying to use PIO remote to build a simple esp8266/Arduino project, it fails with 
"Error: Could not find LD script"

#### Steps to Reproduce

1. Start new project, with wemos d1 mini pro, blink project
2. Try to compile for a remote target, it fails with "Error: Could not find LD script"
3. Force a remote build with --force-remote, this works and is meant to fix 2. above
4. Try another compile for the remote target and it still fails with "Error: Could not find LD script"

### Actual Results
Step 2 above https://pastebin.com/ASMNQxCT
Step 3 above https://pastebin.com/e6RMXSKu (this appears to fail but it's because the serial upload fails, the board isn't connected)
Step 4 above https://pastebin.com/p7Awikd1

### Expected Results
It should work :-)

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
;PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:d1_mini_pro]
platform = espressif8266
board = d1_mini_pro
framework = arduino
upload_port=/dev/ttyUSB0

```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/Blink
*/

void setup(void);
void loop(void);

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(13, OUTPUT);
  Serial.begin(115200);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  Serial.println("BLINK");
  delay(1000);                       // wait for a second
  digitalWrite(13, LOW);
  Serial.println("BLONK");// turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second

}
```

### Additional info
#2848 
#1050

lakidd, 2019-08-23:
Just adding to this, tried on a vanilla MacOS with a fresh PIO install, it was connecting to the same remote and got the identical result

lakidd, 2019-08-27:
> It seems like a known issue when Python 2 and 3 were used on different
> ends. Is it true for your case?

No I'm python 2.7 everywhere from what I can see ?
python3 may be on the machine but the 'python' command drops me into a
python 2.7 shell

ivankravets, 2019-08-27:
It seems like a known issue when Python 2 and 3 were used on different ends. Is it true for your case?

ivankravets, 2019-08-27:
Thanks, I'll re-test it!

ivankravets, 2019-09-02:
Duplicate of https://github.com/platformio/platform-espressif8266/issues/165

We will release a new version of Espressif 8266 dev/platform soon. You will need to update it on a remote machine.

closed 2019-09-02 by ivankravets