platformio/platformio-core #2934: Problem with change default location

Timka654, 2019-08-21:
I'm have problem with russian character, default directorory is `D:\User\OneDrive\Документы\PlatformIO\Projects`
command
`platformio` settings set projects_dir E:\STM\
return error

> The new value for the setting has been set!
> Name                            Value [Default]   Description
> -------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 
> Error: Traceback (most recent call last):
>   File "c:\users\relif\.platformio\penv\lib\site-packages\platformio\__main__.py", line 88, in main
>     cli(None, None, None)
>   File "c:\users\relif\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
>     return self.main(*args, **kwargs)
>   File "c:\users\relif\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
>     rv = self.invoke(ctx)
>   File "c:\users\relif\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 37, in invoke
>     return super(PlatformioCLI, self).invoke(ctx)
>   File "c:\users\relif\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "c:\users\relif\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "c:\users\relif\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>   File "c:\users\relif\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
>     return callback(*args, **kwargs)
>   File "c:\users\relif\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
>     return f(get_current_context(), *args, **kwargs)
>   File "c:\users\relif\.platformio\penv\lib\site-packages\platformio\commands\settings.py", line 72, in settings_set
>     ctx.invoke(settings_get, name=name)
>   File "c:\users\relif\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
>     return callback(*args, **kwargs)
>   File "c:\users\relif\.platformio\penv\lib\site-packages\platformio\commands\settings.py", line 52, in settings_get
>     _defvalue_str = str(_data['value'])
> 
> ============================================================
> 
> An unexpected error occurred. Further steps:
> 
> * Verify that you have the latest version of PlatformIO using
>   `pip install -U platformio` command
> 
> * Try to find answer in FAQ Troubleshooting section
>   https://docs.platformio.org/page/faq.html
> 
> * Report this problem to the developers
>   https://github.com/platformio/platformio-core/issues
> 
> ============================================================

why you not using UTF-8 encoding in IDE? it's helped escape all problems for users from other regions

PlatformIO, version 4.0.0

ivankravets, 2019-08-22:
Thanks for the report! The final release of PIo Core 4.0.1 is coming today.

closed 2019-08-22 by ivankravets

Marooned-MB, 2020-11-23:
I know this is an old and closed issue but it seems to be not working again.
Pasting
`pio settings set projects_dir "c:\Documents and Settings\Marooned\Desktop\Dropbox\różne\projekty\PlatformIO\"`
in terminal results in
`pio settings set projects_dir "c:\Documents and Settings\Marooned\Desktop\Dropbox\róne\projekty\PlatformIO\"`
There is no way to paste Polish character `ż`. Why even all this circus with terminal settings instead of simple `project folder` in the settings? I wanted to switch from archaic Arduini IDE to some modern IDE but such issues looks like pre alpha version :(

commit 0e56a155 (2019-08-22): Fixed an issue when printing settings and file path contains non-ASCII chars // Resolve #2934