platformio/platformio-core #2936: lib folder not includes in CI build

saloid, 2019-08-22:
### Configuration

**Operating system**: GitLabCI

**PlatformIO Version** (`platformio --version`): 4.0.1

### Description of problem
In CI build on GitLab, lib folder not includes and project can`t be build without libs in it. When i am building project in Visual Studio Code - all works.

#### Steps to Reproduce

1. Create any project and place any library in lib folder (lib/libraryName/libraryName.h)
2. Try to build in Visual Studio Code - success
3. Try GitLab-Ci - fail

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp8285]
platform = espressif8266
board = esp8285
framework = arduino
lib_deps = https://github.com/knolleary/pubsubclient, WifiManager
```

**The content of `.gitlab-ci.yml`:**
```yml
image: python:2.7

stages:
 - test

before_script:
  - "pip install -U platformio"

job:
  stage: test
  script: "platformio ci --board=esp8285"
  variables: {PLATFORMIO_CI_SRC: "src/"}
```

### Additional info
lib folder:
```
+---lib
|   |   README
|   |
|   +---fifoBuffer
|   |       fifoBuffer.cpp
|   |       fifoBuffer.h
|   |
|   +---SofSpiSlave
|   |       SoftSpiSlave.cpp
|   |       SoftSpiSlave.h
```

<details>
  <summary>GitLab CI build log</summary>
  
```
Running with gitlab-runner 11.8.0 (4745a6f3)
  on ****** gitlab runner on dev server hcfowzDG
Using Docker executor with image python:2.7 ...
Pulling docker image python:2.7 ...
Using docker image sha256:d75b4eed9ada21ac6b19223a95d56113b54b244aa64848559f146c8c829fb4e8 for python:2.7 ...
Running on runner-hcfowzDG-project-93-concurrent-0 via d45f70928b63...
Fetching changes...
HEAD is now at 0f91747 : denoting in CI_SRC
From https://git.*******
[... 98 lines cut ...]
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
Compiling .pio/build/esp8285/src/main.cpp.o
Generating LD script .pio/build/esp8285/ld/local.eagle.app.v6.common.ld
src/main.cpp:1:26: fatal error: SoftSpiSlave.h: No such file or directory
</details>

**********************************************************************
* Looking for SoftSpiSlave.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SoftSpiSlave.h"
* Web  > https://platformio.org/lib/search?query=header:SoftSpiSlave.h
*
**********************************************************************

 #include "SoftSpiSlave.h"
                          ^
compilation terminated.
*** [.pio/build/esp8285/src/main.cpp.o] Error 1
========================= [FAILED] Took 10.50 seconds =========================
ERROR: Job failed: exit code 1
```

saloid, 2019-08-22:
Same issue: https://stackoverflow.com/questions/44465254/platformio-ci-libraries-not-found

ivankravets, 2019-08-23:
https://stackoverflow.com/questions/44465254/platformio-ci-libraries-not-found this is an old issue, I can't reproduce it.

ivankravets, 2019-08-23:
In your case, 

1) If you have already pre-configured PlatformIO project, please use `pio run` instead. See example https://github.com/platformio/platform-espressif32/blob/develop/.travis.yml#L27

2) If you want to test explicitly source code, you can create a temporary `platformio.ini` and pass it to CI command or use `--project-option`. See http://docs.platformio.org/en/latest/userguide/cmd_ci.html

closed 2019-08-23 by ivankravets