platformio/platformio-core #2941: local libraries and include dependencies

MarioJose, 2019-08-23:
### Configuration

Atom 1.40.1 x64
PlatformIO 2.2.0 Core 4.0.2
GNU/Linux Debian Streach x64 (9)

### Description of problem
Dependency error compiling code when not include custom lib in main code. Including myclim.h is optional.

I tried to compile this code in Arduino IDE, and have no error, including or not myclim.h in main.cpp. In this case, DHT.h is always compiled in Arduino IDE, even if myclim.h is not included in main.cpp.

#### Steps to Reproduce
1. Compile code with #include <myclim.h> in main.cpp (not error when compiling)
2. Compile code without #include <myclim.h> in main.cpp (error when compiling)

### Actual Results
If myclim.h is included in main code, it is compiled without errors. But, if I remove the line #include <myclim.h> in main.cpp, I get the follow error:

```
Processing pro16MHzatmega328 (platform: atmelavr; board: pro16MHzatmega328; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/pro16MHzatmega328.html
PLATFORM: Atmel AVR > Arduino Pro or Pro Mini ATmega328 (5V, 16 MHz)
HARDWARE: ATMEGA328P 16MHz 2KB RAM (30KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf

LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 17 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <mylib>
Compiling .pioenvs/pro16MHzatmega328/src/main.cpp.o
Compiling .pioenvs/pro16MHzatmega328/lib2cc/mylib/myclim.cpp.o
In file included from lib/mylib/myclim.cpp:1:0:
lib/mylib/myclim.h:5:17: fatal error: DHT.h: No such file or directory

*************************************************************
* Looking for DHT.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:DHT.h"
* Web  > https://platformio.org/lib/search?query=header:DHT.h
*
*************************************************************

compilation terminated.
*** [.pioenvs/pro16MHzatmega328/lib2cc/mylib/myclim.cpp.o] Error 1
```

**The content of `platformio.ini`:**
```ini
[env:promini]
platform = atmelavr
board = pro16MHzatmega328
framework = arduino
upload_port = /dev/ttyUSB*
```

**Source file to reproduce issue:**

/lib/mylib/mylib.h

```c
#ifndef MYLIB_H
#define MYLIB_H

#include <Arduino.h>

void fn();

#endif
```

/lib/mylib/mylib.cpp

```c
#include <mylib.h>

void fn()
{
  Serial.println("fn1");
}
```

/lib/mylib/myclim.h

```c
#ifndef MYCLIM_H
#define MYCLIM_H

#include <Arduino.h>
#include <DHT.h>

void fnclim();

#endif
```

/lib/mylib/myclim.cpp

```c
#include <myclim.h>

void fnclim()
{
  Serial.println("fnclim");
}
```

/src/main.cpp

```c
#include <Arduino.h>
#include <mylib.h>
#include <myclim.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

ivankravets, 2019-08-23:
> In this case, DHT.h is always compiled in Arduino IDE

because you installed it before into the global Arduino scope? Just add to your `platformio.ini`
```
[env:promini]
platform = atmelavr
board = pro16MHzatmega328
framework = arduino
upload_port = /dev/ttyUSB*
lib_deps = 
  DHT sensor library
```

See

------

*************************************************************
* Looking for DHT.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:DHT.h"
* Web  > https://platformio.org/lib/search?query=header:DHT.h
*
*************************************************************

closed 2019-08-23 by ivankravets

pfeerick, 2019-08-23:
The Arduino IDE will fail also if DHT library is not installed, so PlatformIO is behaving exactly the same way. Commenting out the `#include <myclim.h>` line in the main file will result in an error on the Arduino IDE if the DHT library is not installed. Even though the compiler on the Arduino IDE doesn't end up using that file in the final build, it still seems to process the include statements in it and consequently fails with a 

```
In file included from sketch/myclim.cpp:1:0:
myclim.h:5:17: error: DHT.h: No such file or directory
compilation terminated.
exit status 1
DHT.h: No such file or directory
```

MarioJose, 2019-08-24:
DHT is installed in both system (in Arduino IDE path and in PlatformIO path). As I said, when I have  `#include <myclim.h>` in main.cpp, the compilation is Ok. I get error only when I remove `#include <myclim.h>`  in main.cpp. I think that if I do not use myclim.h, and consequently dht.h (no including `#include <myclim.h>`  in main.cpp), PlatformIO will not search for dht.h.

pfeerick, 2019-08-24:
The adafruit DHT library has a dependency on the Adafruit Unifed sensor library... have you got that installed also?

```
lib_deps = 
  Adafruit Unified Sensor
  DHT sensor library
```

MarioJose, 2019-08-24:
It is not problem of DHT. If I change dht.h for RTClib.h, I get the same error.

pfeerick, 2019-08-24:
See what happens with the code [in this branch](https://github.com/pfeerick/broken-code/tree/myclim)... either I missed something when replicating the code, or something is different on your end. It compiled without any errors, with `myclim.h` commented out. 

You'll be able to download it as a zip via https://github.com/pfeerick/broken-code/archive/myclim.zip

MarioJose, 2019-08-24:
works only if I add this in platformio.ini
lib_deps = 
  DHT sensor library

pfeerick, 2019-08-24:
That is the correct behaviour. 

Regardless of if you actually include/use the header that references `DHT.h`, that file is still processed by the compiler, and hence must exist. So it must be installed. It is only later in in the compile process that it is dropped. 

This is exactly the same behaviour as the Arduino IDE. If you remove the DHT library, and  comment out `#include <myclim.h>` in your main sketch file... you'll get the error I gave above. If you install the DHT library, the error will go away - even though the library isn't actually used. By altering the `lib_deps` line in `platformio.ini` you are telling platformio to install the library if it's not installed, as well as that it's a library needed by the project at some point.

MarioJose, 2019-08-25:
Hi, @pfeerick. Excuse me, I think that I'm not understand something because my english.

The library DHT or any other that I use in myclim.h **is installed**, both in Arduino library path and PlatformIO library path. Then, the library exists. It makes sense that I not have a compiling error when include myclim in main source. But, why PlatformIO return error that not found library (when not include it main source) if library file exists and is installed? 

If I don't use myclim, not make sense compiling DHT and make final binary bigger than necessary (Arduino is limited in space). Maybe I'm not know the exactly process of compiling.

If I understand the output, PIO (PlatformIO) first pre compiling libraries included in main.cpp. Then DHT is pre compiled before compiling myclim (whem myclim is included in main.cpp). When I not include myclim in main cpp, PIO try compiling myclim and because DHT is not pre compiled, PIO return error. I think that PIO must search for dependency in local library. Then, PIO must "see" DHT in myclim, pre compiling it before compiling myclim. But the question is, why compiling myclim if I not using it when not include it in main.cpp?