platformio/platformio-core #2947: Add support for rust

abelykh0, 2019-08-26:
What kind of issue is this?

- [ ] **Feature Request**.
Add support for rust and cortex-m-rtfm
https://github.com/japaric/cortex-m-rtfm

ivankravets, 2020-06-05:
Related links:
- https://community.platformio.org/t/how-would-i-implement-a-rust-compiler-that-links-with-any-framework/14113
- https://github.com/dzervas/platformio-arduino-rust

dzervas, 2020-06-25:
I've created a POC on the above linked repo, take a look. Shouldn't be too hard to use my example for another MCU

mash-graz, 2021-02-13:
better rust support in pio would be really appreciable!

perhaps it could also make some sense to combine this improvements with better [rt-thread](https://www.rt-thread.io/) support, because this particular RTOS and its middleware packages also utilize the python based SCons build system, and there are also some rust integration attempts available for it ([rttrust](https://crates.io/crates/rttrust)), which look very similar to the solutions suggested here.

CoderScribe, 2021-04-17:
I'd also like to see multi-compiler (and testing framework) support, there's a definite growing trend of using Rust in embedded, it's even starting to standardise its own CMSIS-like approach which chimes well with Platform IO's goals. It has also recently been proposed as the Linux kernel's second official language (with Rust growing even faster in the systems programming space). There's something to be said for being able to have a single language for writing safe code at all these various levels, especially in the IoT space where development can often be full stack. It would bridge that final gap in providing a unified solution across the embedded space (and maybe even a bit more).

HKalbasi, 2021-04-19:
What is the path toward this goal? How we can contribute?

ivmarkov, 2021-06-20:
Shameless plug: here's [my effort](https://github.com/ivmarkov/cargo-pio). Currently used to drive [this Rust binary crate project](https://github.com/ivmarkov/rust-esp32-std-hello) with PlatformIO.

dzervas, 2021-06-29:
@HKalbasi You can contribute by creating Rust bindings for all of the C++ code. Maybe https://cxx.rs can help. There's a blog post of what I've done (more or less): https://whynot.fail/coding/rusty-arduino-bindings/

It's a huge undertaking, mostly due to the complexity of the C++ code and its build system (it's really, REALLY hard for me to follow which code is actually called with all of the conditional compilation flags)

What could be done is to start a dedicated project for each architecture/MCU with (at least partially) manually written bindings.