platformio/platformio-core #2971: KeyError when trying to access env['PIOHOME_DIR']

Pharap, 2019-09-02:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 8.1

**PlatformIO Version** (`platformio --version`): PlatformIO 4.0.3

### Description of problem

Environment key `'PIOHOME_DIR'` does not exist when running a 'pre' script.

#### Steps to Reproduce

1. Add the following to `platformio.ini`:
```ini
extra_scripts =
 pre:test.py
```
2. Create `test.py` with the following contents:
```python
Import("env")

print env['PIOHOME_DIR']
```

### Actual Results

Console gives errors:
```
KeyError: 'PIOHOME_DIR':
  File "C:\users\user\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 124:
    env.SConscript(item, exports="env")
  File "I:\User\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "I:\User\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "I:\Documents\Github\project\test.py", line 3:
    print env['PIOHOME_DIR']
  File "I:\User\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 410:
    return self._dict[key]
```

### Expected Results

Console should output either:
```
C:\users\user\.platformio
```
Or:
```
I:\User\.platformio
```

\* `C:\users\user\.platformio` is a symlink to `I:\User\.platformio`

ivankravets, 2019-09-02:
`pio run -t envdump`

See valid variables

ivankravets, 2019-09-02:
You need https://github.com/platformio/platformio-core/blob/1be2e510da1e20d37c4ac13a7259ee01fe4c9e2e/platformio/builder/main.py#L61

closed 2019-09-02 by ivankravets

Pharap, 2019-09-02:
Do you happen to know when this change was made?
I couldn't find any mention of the environment variable names changing in the changelog.

ivankravets, 2019-09-02:
These are internal variables and we use them for own needs. I recommend using PIO Core API instead. 
```python
platformio.project.helpers import get_project_core_dir

print(get_project_core_dir())
```

Please note that Core folder can differ between projects.