platformio/platformio-core #2973: Build Pre-hooks not getting called

guru-florida, 2019-09-03:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Linux amd64 (Ubuntu 19)

**PlatformIO Version** (`platformio --version`): 4.0.2

### Description of problem

Attaching extra_script hooks using pre: does not execute pre buildproj hooks as described in docs:
http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#before-pre-and-after-post-actions

#### Steps to Reproduce

1. Create extra_script.py next to platformio.ini
2. In platformio.ini, add extra_scripts = pre:extra_script.py to your active env: section

#### extra_script.py
```
Import("env")

env.Append(CPPDEFINES=("FOO=1"))

def define_sensor_drivers(source, target, env):
    print("Configuring sensor device drivers...")
    env.Append(CPPDEFINES=("BAR=1"))

env.AddPreAction("buildprog", define_sensor_drivers)
```

### Actual Results

* The FOO=1 does get added to compile lines (therefor pre: script is getting run)
* Neither "Configure sensor device drivers..." is printed, nor is BAR added to compile defs

### Expected Results

env.AddPreAction("buildprogs", callback...) should call the define_sensor_drivers function.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
default_envs = nodemcuv2

..snip...

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
extra_scripts = pre:extra_targets2.py
```

### Additional info

If the pre: is removed from extra_scripts line the method does get run, but only after the compilation is done.

There was also some other tickets related to this that didnt seem to be resolved.

ivankravets, 2019-09-03:
See docs about extra script launch types http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#launch-types

You use PRE, which means to call your script BEFORE a build system starts. So, at this point there is no information about any targets. This PRE type is useful when you would like to do some PRE manipulation before the build process starts.

I changed `extra_scripts = pre:extra_targets2.py` to `extra_scripts = post`:extra_targets2.py` and it works.

P.S: What is your use case? Maybe, adding action directly on physical targets is better. I mean `env.AddPreAction("$BUILD_DIR/${PROGNAME}.elf", [callback1, callback2,...])`. `buildprog` is a target when ALL files are compiled and PROGRAM is ready.

closed 2019-09-03 by ivankravets

guru-florida, 2019-09-03:
@ivankravets I need the build step to run *before* the compilation begins so I can process a config file and add CPP DEFINES. 

I have a project that loads a number of dependent sensor libraries. Other users may only want to add support for fewer sensors and by having the build hook I can process the user's requested setup and turn on/off includes/etc and therefor support for that sensor. Setup may be in platformio.ini or in a seperate csv-like file. (not sure yet)

I believe changing pre: to post: has it run after compilation, yes? As mentioned in the Additional Info it runs the script if pre: is just removed all together (which I think is equivalent to post:) but this is too late to add CPP_DEFINES. The extra_targets2.py was a cut-paste typo. 

I also tried attaching the pre: to my main object file $BUILD_DIR/Nimble.cpp.o but that doesnt hook either.

ivankravets, 2019-09-03:
Just use bare `pre:myscript.py` and `env.Append(CPPDEFINES=[(macroname, value), ...()]). No need to use yet another AddPreAction

guru-florida, 2019-09-03:
Ok! Now I get it. I was confusing the pre: and post: with the env.AddPreAction() when they are really not that related. Having the pre: script just run without using AddPreAction will work great.

Once I changed pre: to post: and used this line I was able to generate the CPP defines immediately before my main source file compiled....so this would work too but I like the top method.
```
env.AddPreAction("${BUILD_DIR}/src/main.cpp.o", define_sensor_driver)
```

Thanks for the help!!