platformio/platformio-core #2985: arduino nano 33 ble sense

jpw7792, 2019-09-09:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [X ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem
need arduino nano 33 ble sense added to platformio board list

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

ivankravets, 2019-09-09:
Duplicate of https://github.com/platformio/platform-nordicnrf52/issues/53

closed 2019-09-09 by ivankravets

jpw7792, 2019-09-09:
> Duplicate of platformio/platform-nordicnrf52#53

Ivan:

Forgive me for being an arduino newbie, but I'm wondering when I can expect
to be able to "select" the ArduinoNano33BLESense board within the
Atom/PlatformIO IDE ?

I've thus far figured out that the "platform" for the ArduinoNano33BLESense
(board not selectable so far) is "nRF52", correct ?
And, the "framework" is going to be Arduino and/or Mbed OS, correct ?

Any advice about how I can get a jumpstart using the ArduinoNano33BLESense
with ready-to-go sensor examples would be much appreciated.

Thank you,

Jeff Wagner