platformio/platformio-core #3022: Booleans in platformio.ini not parsed properly

InvncibiltyCloak, 2019-09-15:
What kind of issue is this?
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

**Operating system**: Ubuntu 19.04

**PlatformIO Version**: Version 4.0.3

### Description of problem
I am attempting to override some of the options for my board, as described in https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#override-board-configuration

The option I want to override is a boolean in the JSON. I am trying to override 'upload.use_1200bps_touch'. However, any value I enter into the platformio.ini file gets parsed as a string (not a boolean), and Python considers any non-empty string as True. Therefore it is nearly impossible to override this (or any other boolean) with the value False. The entry needs to be properly parsed to a boolean.

#### Steps to Reproduce
1. Create platform.ini file for a board that has upload.use_1200bps_touch set to True by default.
2. Add the option "board_upload.use_1200bps_touch = False" to platform.ini

### Actual Results
The upload script still executes the use_1200bps_touch code, because False was parsed as a string which evaluates to True. You will see in the terminal "Forcing reset using 1200bps open/close on port" when this option executes.

### Expected Results
False should be False. There should be no text "Forcing reset using 1200bps open/close on port" in the terminal.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:a-star32U4]
platform = atmelavr
board = a-star32U4
board_upload.use_1200bps_touch = False
```

InvncibiltyCloak, 2019-09-15:
As a note, a workaround to this problem is to set the value to nothing, which is parsed as None, which evaluates to False in the code.

Example:
```ini
[env:a-star32U4]
platform = atmelavr
board = a-star32U4
board_upload.use_1200bps_touch = 
```

ivankravets, 2019-10-02:
Living an empty value is a true temporary solution here.

commit 3177aaf5 (2019-10-24): Fixed an issue when booleans in "platformio.ini" are not parsed properly // Resolve #3022
closed 2019-10-24 by ivankravets