platformio/platformio-core #3025: build "hang" while installing GIT based platform/frameworks.

atanisoft, 2019-09-16:
What kind of issue is this?

- [x] **Feature Request**.

When building a project for the first time and the dependent platform(s) are not installed the console output will appear to "hang" with the following output:
```
> Executing task in folder ESP32CommandStation: .....\platformio.exe run --environment pcb_oled_sh1106 <

Processing pcb_oled_sh1106 (platform: https://github.com/platformio/platform-espressif32.git; board: esp32dev; framework: espidf, arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
```
This project is using the GIT versions for arduino-esp32 and esp-idf via the following platformio.ini overrides:
```
[env]
# overrides to use GIT versions rather than PIO cached copies.
platform_packages =
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
  framework-espidf @ https://github.com/espressif/esp-idf.git#release/v3.2
```

ivankravets, 2019-10-02:
Could you provide a simple project to reproduce this?

@valeros please check.

valeros, 2019-10-02:
Hi @atanisoft ! Tried a simple project with your config and didn't notice any problem. Maybe it's downloading frameworks when it hangs?

atanisoft, 2019-10-02:
@ivankravets any of the example projects will reproduce it.

@valeros that might be the case. It seems to be more of a problem when using esp-idf from git which has a lot of submodules etc which take time to init.

ivankravets, 2019-10-04:
@valeros do you see Git cloning progress? I don't remember how do we run it.

atanisoft, 2019-10-04:
When starting from a "fresh" tree (no frameworks/platforms installed) and using the git references, I've seen the git clone process only once that I can remember. Often it "hangs" until it has the necessary bits installed and then starts the clone process etc.

ivankravets, 2019-10-24:
Did you resolve this issue?

atanisoft, 2019-10-24:
The latest alpha of PIO Core seems to have helped some, it now at least prints that it is installing the platform (via git) but output is still blank for an extended period (somewhat expected):
```
> Executing task in folder ESP32CommandStation: C:\Users\mdunston\.platformio\penv\Scripts\platformio.exe run <

Processing esp32 (platform: https://github.com/platformio/platform-espressif32.git; board: esp32dev; framework: espidf, arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PlatformManager: Installing platform-espressif32
espressif32 @ 31f8624 is already installed
The platform 'https://github.com/platformio/platform-espressif32.git' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
```
After this output it is blank, but frameworks are being installed via git references. It would be good to show the git clone process output in the console as an indicator to the end user that it is actively doing something and not hung.

ivankravets, 2019-10-24:
I still can't reproduce it. Could you make a video with the issue? See my screenshot:

[image]

atanisoft, 2019-10-24:
AVR is a small repo, use esp32 with IDF pointing to GIT using:
```
platform_packages =
  framework-espidf @ https://github.com/espressif/esp-idf.git#release/v3.3
```
You  might be able to reproduce this same behavior by overriding framework-arduino-avr to a GIT version but it likely is still smaller than ESP-IDF which has a number of dependencies via submodules.

ivankravets, 2019-10-24:
It works too. Let's keep this issue open for a while maybe someone provides more info on how to reproduce it.
[image]

atanisoft, 2019-10-24:
Hmm, this could be a case of windows vs linux/mac. Your output looks like I'd expect but I don't see the same on windows.

ivankravets, 2019-10-24:
What is your version of Git client?

atanisoft, 2019-10-24:
```
$ git version
git version 2.19.1.windows.1
```
It isn't the latest (2.23.0) which I'm updating to now.

ivankravets, 2019-10-24:
@valeros could you update Git to the latest version. Does it work for you as in my case on macOS?

Use this for test
```
[platformio]
core_dir = C:/tmp-wiring-pio-core

[env:uno]
platform = atmelavr
platform_packages =
  framework-espidf @ https://github.com/espressif/esp-idf.git#release/v3.3
framework = arduino
board = uno
```

atanisoft, 2019-10-24:
git 2.23.0 behaves similarly to previous installed version.

ivankravets, 2020-02-14:
Do we have any updates on this issue? Could we close it?

atanisoft, 2020-02-14:
I've stopped using PIO since moving over to Linux as my primary OS. There is something in the esp-idf builder that breaks fully and I couldn't find it. So now I've shifted over to using cmake instead.

closed 2020-02-14 by atanisoft

ivankravets, 2020-02-15:
We work on adding support for ESP-IDF 4.0. So, we would be happy to see you again on our bright side and sorry for the issue.

atanisoft, 2020-02-15:
@ivankravets When IDF v4 is ready I'll give it a try. I'm currently using idf.py to build as the VSCode plugin from Espressif has some sort of issue related to Python (reproduced on windows and linux so far, issue TBD).

ivankravets, 2020-02-15:
Which interesting features that plugin has over PlatformIO? Can both plugins work together?

atanisoft, 2020-02-15:
I believe the two plugins can work together but they may have conflicts in the .vscode directory for paths since IDF usually resides in /home/<user>/esp or similar path.

The plugin has three advantages over PIO currently:
1. Native CMake integration, same as running idf.py <target> which invokes CMake.
2. Graphical "menuconfig" within VSCode.
3. Easy switching of IDF versions (git checkout -- <tag>) without needing to upgrade/downgrade the plugin to ensure the build works.

Since my project no longer has Arduino dependencies, it makes sense for me to use the lower level IDF approach.

ivankravets, 2020-02-15:
Thanks! A few good things:

1. New integration for ESP-IDF 4 will be based on the CMake generator. So, we will use CMake to get a list of build items and use the PlatformIO Build system for the rest tasks. We already did this for Zephyr RTOS. It means that you can anytime build the same project with idf.py or PlatformIO.
2. "menuconfig" - yes, there is a feature request and we will try to resolve it soon.
3. We currently support only stable software. However, if we will support CMake provided information, it should be possible to use dev-version of ESP-IDF too. It depends on ESP-IDF build process and that all steps are declared in CMake. Otherwise, we will not know how to build some targets which are no declared in CMake.