platformio/platformio-core #3030: Home: Could not update platform

jmrg-link, 2019-09-18:
PIO Core Call Error: "Platform C:\\Users\\tekno\\AppData\\Roaming\\SPB_Data\\.platformio\\platforms\\nordicnrf52\r\n--------\nUpdating nordicnrf52                     @ 3.7.0          [Up-to-date]\nUpdating tool-jlink                      @ 1.63208.0      [1.65001.0]\r\nUninstalling tool-jlink @ 1.63208.0: \t[OK]\nPackageManager: Installing tool-jlink @ 1.65001.0\r\nDownloading...\nUnpacking...\ntool-jlink @ 1.65001.0 has been successfully installed!\nUpdating toolchain-gccarmnoneeabi        @ 1.70201.0      [Up-to-date]\nUpdating framework-arduinonordicnrf5     @ 1.600.190830   [Up-to-date]\n\n\nError: Detected unknown package 'C:\\Users\\tekno\\AppData\\Roaming\\SPB_Data\\.platformio\\packages\\tool-openocd @ ~2.1000.0'"

ivankravets, 2019-10-02:
Please remove `C:\Users\tekno\AppData\Roaming\SPB_Data\.platformio\packages\tool-openocd**` folders and restart

closed 2019-10-02 by ivankravets

jmrg-link, 2019-11-15:
Thanks.
Platform run correctly