platformio/platformio-core #3034: Configuration file parameters partly ignored when using --project-conf

daniel-starke, 2019-09-18:
**PlatformIO Core**
------------------------------------------------------------------

### Configuration

**Operating system**: Windows 7

**PlatformIO Version** (`platformio --version`): 4.0.3

### Description of problem
platformio section is not evaluated if a non-default configuration file is provided with `-c`.

#### Steps to Reproduce

1. `platformio run -c test.ini`

### Actual Results

.pio is used as default.

### Expected Results

The path defined in workspace_dir is used.

### Problems with PlatformIO Build System:

**The content of `test.ini`:**
```ini
[platformio]
workspace_dir = bin

[env:wavrec]
platform = espressif32
framework = arduino
board = wemosbat
monitor_speed = 115200
```

**Source file to reproduce issue:**
```cpp
setup(void) {}
loop(void) {}
```

### Additional info

Defining the environment variable PLATFORMIO_WORKSPACE_DIR works regardlessly.

ivankravets, 2019-09-27:
Oh, this issue took a lot of efforts, but it worth it. See https://github.com/platformio/platformio-core/commit/d2abac9b18ff47cd2ab32312995ba52e78cfd62a

Could you re-test with `pio upgrade --dev`?

closed 2019-09-27 by ivankravets

daniel-starke, 2019-09-30:
Thank you. Works for me as expected.

ivankravets, 2019-09-30:
Thanks for the report!

Happy coding with PlatformIO!

daniel-starke, 2019-09-30:
Hm... would had been nice to have a way to "upgrade" back to stable...

ivankravets, 2019-09-30:
`pip uninstall platformio`
`pip install platformio`

commit d2abac9b (2019-09-27): Fixed an issue when configuration file options partly ignored when ``--project-conf`` // Resolve #3034 (#3055)

* Fixed an issue when configuration file options partly ignored when using custom ``--project-conf`` // Resolve #3034

* Py2 compatible makedirs

* Fix circle dependency

* Fix broken import in test examples

* Fix history

* Remove YAPF markers

* PyLint fix

* Fix invalid project conf path

* Move PIO Core to the root on Windows, issue with long CPPPATHs

* Respect global PLATFORMIO_BUILD_CACHE_DIR env var

* Fix Appveyor paths

* Minor changes