platformio/platformio-core #3040: Home: Libraries: Could not install library

tamagoz, 2019-09-22:
PIO Core Call Error: "Library Storage: C:\\Users\\GreatODev\\.platformio\\lib\nLibraryManager: Installing id=1747 @ ^1.0.4\nDownloading...\nUnpacking...\nESP8266MQTTMesh @ 1.0.4 has been successfully installed!\nInstalling dependencies\nLooking for ESPAsyncTCP library in registry\r\nFound: https://platformio.org/lib/show/305/ESPAsyncTCP\nLibraryManager: Installing id=305 @ ^1.1.3\r\nDownloading...\nUnpacking...\nESPAsyncTCP @ 1.2.1 has been successfully installed!\nLooking for AsyncMqttClient library in registry\r\nFound: https://platformio.org/lib/show/346/AsyncMqttClient\nLibraryManager: Installing id=346 @ ^0.8.1\r\nDownloading...\nUnpacking...\nAsyncMqttClient @ 0.8.2 has been successfully installed!\nInstalling dependencies\nLibraryManager: Installing id=305 @ ^1.1.0\r\nESPAsyncTCP @ 1.2.1 is already installed\nLooking for AsyncTCP library in registry\r\nFound: https://platformio.org/lib/show/1826/AsyncTCP\nLibraryManager: Installing id=1826 @ ^1.0.0\r\nDownloading...\nUnpacking...\nAsyncTCP @ 1.1.0 has been successfully installed!\nLibraryManager: Installing id=1826\nAsyncTCP @ 1.1.0 is already installed\nLooking for Ticker-esp32 library in registry\r\nFound: https://platformio.org/lib/show/1938/Ticker-esp32\nLibraryManager: Installing id=1938\nDownloading...\nUnpacking...\nTicker-esp32 @ 1.1.0 has been successfully installed!\nLibraryManager: Installing AsyncMqttClient\n\n\nError: Traceback (most recent call last):\n  File \"c:\\users\\greatodev\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 90, in main\n    cli(None, None, None)\n  File \"c:\\users\\greatodev\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 764, in __call__\n    return self.main(*args, **kwargs)\n  File \"c:\\users\\greatodev\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 717, in main\n    rv = self.invoke(ctx)\n  File \"c:\\users\\greatodev\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 37, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"c:\\users\\greatodev\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\greatodev\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\greatodev\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 956, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"c:\\users\\greatodev\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 555, in invoke\n    return callback(*args, **kwargs)\n  File \"c:\\users\\greatodev\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 17, in new_func\n    return f(get_current_context(), *args, **kwargs)\n  File \"c:\\users\\greatodev\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib.py\", line 150, in lib_install\n    force=force)\n  File \"c:\\users\\greatodev\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\lib.py\", line 388, in install\n    force=force)\n  File \"c:\\users\\greatodev\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\lib.py\", line 359, in install\n    force=force)\n  File \"c:\\users\\greatodev\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 701, in install\n    track=True)\n  File \"c:\\users\\greatodev\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 500, in _install_from_url\n    return self._install_from_tmp_dir(_tmp_dir, requirements)\n  File \"c:\\users\\greatodev\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 583, in _install_from_tmp_dir\n    shutil.move(tmp_dir, pkg_dir)\n  File \"C:\\Users\\GreatODev\\.platformio\\python27\\Lib\\shutil.py\", line 300, in move\n    rmtree(src)\n  File \"C:\\Users\\GreatODev\\.platformio\\python27\\Lib\\shutil.py\", line 247, in rmtree\n    rmtree(fullname, ignore_errors, onerror)\n  File \"C:\\Users\\GreatODev\\.platformio\\python27\\Lib\\shutil.py\", line 247, in rmtree\n    rmtree(fullname, ignore_errors, onerror)\n  File \"C:\\Users\\GreatODev\\.platformio\\python27\\Lib\\shutil.py\", line 247, in rmtree\n    rmtree(fullname, ignore_errors, onerror)\n  File \"C:\\Users\\GreatODev\\.platformio\\python27\\Lib\\shutil.py\", line 252, in rmtree\n    onerror(os.remove, fullname, sys.exc_info())\n  File \"C:\\Users\\GreatODev\\.platformio\\python27\\Lib\\shutil.py\", line 250, in rmtree\n    os.remove(fullname)\nWindowsError: [Error 5] Access is denied: 'C:\\\\Users\\\\GreatODev\\\\.platformio\\\\lib\\\\_tmp_installing-o7r53g-package\\\\.git\\\\objects\\\\pack\\\\pack-5f1a488ae4a245adf786bcf3a6619f0d17e356b8.idx'\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

ivankravets, 2019-10-02:
> [Error 5] Access is denied: 'C:\\Users\\GreatODev\\.platformio\\lib

Please fix permissions.

closed 2019-10-02 by ivankravets

Bmooij, 2019-10-08:
`shutil.move(tmp_dir, pkg_dir)` can copy the readonly files. But cannot remove them with `shutil.rmtree`.

see: https://stackoverflow.com/questions/21261132/shutil-rmtree-to-remove-readonly-files