platformio/platformio-core #3046: Qt Creator integration enhancement request

dewhisna, 2019-09-25:
What kind of issue is this?

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: All

**PlatformIO Version** (`platformio --version`): 4.0.3

### Additional info

The current integration with Qt Creator works quite well with its ability to output `.pro` files.  However, I would like to see it output it as [Qt Creator generic project](https://doc.qt.io/qtcreator/creator-project-generic.html) files instead, with files like `.config`, `.creator`, `.files`, `.include`, etc.  This will allow it to be used with kit configurations that don't include Qt itself, as `.pro` files are more specific to projects using Qt and requires `qmake` and a full Qt installation.  Plus, `qmake` is a dying application which they are gradually replacing with `cmake`.

All of the content of the `.pro` file that PlatformIO creates fits nicely in these generic project files, which Qt Creator will load together as a project.  Just put the predefined macros (or defines) in the `.config` file.  Put the list of project source files in the `.files` file.  And list the include paths in the `.include` file.  And create a `.creator` file with a single `[General]` line.  And if there are build flags, `.cflags` and `.cxxflags` files can be used.

I think this would be a relatively easy change to PlatformIO, but would be a nice improvement in its export to Qt Creator and will allow people to use it with just Qt Creator installed without installing all of Qt too.

ivankravets, 2019-09-26:
Could you provide PR? See examples https://github.com/platformio/platformio-core/tree/develop/platformio/ide/tpls

The source code of PlatformIO is located in 
```
python -c "import platformio;print(platformio)"
```

dewhisna, 2019-09-30:
PR created...

Since Qt Creator can default to running `make` for generic targets, this new template also adds a `Makefile` that will allow Qt Creator to automatically build the project without having to manually modify the project settings to launch `platformio`.

This `qtcreator-generic` IDE template diverges from the `qtcreator` template in that it doesn't require a Qt Kit to be setup that contains `qmake`.  It also outputs compiler flags so that detail from custom scripts that alter the compiler flags can be captured by the code model in Qt Creator.  It does, however, lose the `HOMEDIR` isolation of include paths that the `qtcreator` template provides -- this is because I see no good way to do it with the generic project mechanism since it's not parsed by `qmake`.

Run with: `platformio init --ide qtcreator-generic`

ecorm, 2020-09-19:
The QtCreator generator should add the following lines to the platformio.pro file:

```
CONFIG += console
CONFIG -= app_bundle
CONFIG -= qt
```

to avoid the need for the user to install the Qt library.

belm0, 2022-05-14:
> The QtCreator generator should add the following lines to the platformio.pro file:
> 
> ```
> CONFIG += console
> CONFIG -= app_bundle
> CONFIG -= qt
> ```
> 
> to avoid the need for the user to install the Qt library.

I tried this, but Qt Creator still says "no suitable kits found".

belm0, 2022-05-14:
revised PR:  #4262

PR #4262 (2022-05-14): qtcreator IDE template now generates a "generic" Qt project
Change the `qtcreator` IDE template to generate a [Qt Creator generic project](https://doc.qt.io/qtcreator/creator-project-generic.html)

(In contrast, the existing `qtcreator` template creates a native project having dependencies on Qt runtime and the `qmake` tool.)

Fixes #3046

Derived from original PR #3072
closed 2022-05-14 by ivankravets