platformio/platformio-core #3050: Header files in `lib_extra_dirs`are not being recognized

jxsl13, 2019-09-25:
What kind of issue is this?

- [x] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: macOS 10.14

**PlatformIO Version** (`platformio --version`): 4.0.3

### Description of problem
[community.platform.io Topic](https://community.platformio.org/t/one-platformio-ini-for-multiple-projects/9716/5?u=jxsl13)

I created a folder called `target` that ought to contain target specific source files and headers.
I wanted to add for every target defined in the `platformio.ini` to link and compile specific files, in order for it to work exactly like the `lib` folder does.

The `target/LPC1768` or `target/ATmega328p` folders also contain header files, that are to being found, even tho in the `lib` folder they are being found.

### Actual Results
Header files referenced with `lib_extra_dirs = target/LPC1768` are not being found, e.g. `target/LPC1768/LPC17xx.h`

### Expected Results
header files being found and linked.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:lpc1768]
platform = nxplpc
board = lpc1768

; Needs to stay for every probe
build_flags = -D MCB1700, -D LPC1768, -Wl,-T"./scripts/target/LPC1768/LPC1768.ld", -D DEBUG
extra_scripts = scripts/target/LPC1768/extra_script.py

; ignore everythin in lib folder and only add the platform specific library
lib_extra_dirs = target/LPC1768

;J-Link Probe
upload_protocol = jlink-jtag
debug_tool = jlink

[env:atmega328p]
platform = atmelavr
board = atmega328p

; atmega328p specific binary files, e.g. board init etc.
lib_extra_dirs = target/ATmega328p

; build and debugging flags
build_flags = -D ATMEGA328P, -D DEBUG

; upload port.
upload_port = /dev/cu.SLAB_USBtoUART```

My repository revision [click me for GitHub repo](https://github.com/jxsl13/lpc1768_blinky_example/tree/9111b27ac1580776882ed38b86508c71f173e7cc)

ivankravets, 2019-10-02:
@valeros please check.

valeros, 2019-10-02:
Hi @jxsl13 ! Each library should be placed in a separate folder so LDF will be able to determine its sources. Change `lib_extra_dirs = target/LPC1768` to `lib_extra_dirs = target` and don't forget to change LDF mode to `lib_ldf_mode = chain+` because you have conditional includes in `proxy.h` file.

closed 2019-10-02 by valeros