platformio/platformio-core #3078: Support for Heltec CubeCell series

SebDominguez, 2019-10-02:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [x] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

I'm requesting [this board](https://heltec.org/project/htcc-ab01/) to be added into PlatformIO.

This board is relatively new and the MCU is low power. It's based on an ASR6501 and unfortunately this platform doesn't exist yet.

Heltec has already made this board [Arduino compatible](https://github.com/HelTecAutomation/ASR650x-Arduino)

Skipper-is, 2019-10-25:
Would be great for this to be on platformio

irukard, 2019-12-16:
Indeed! This is a great, low power dev board with LoRa radio.

jacobsonjn, 2020-02-12:
Also would like this

crapougnax, 2020-02-12:
Looking forward to!

lnlp, 2020-03-05:
@ivankravets This request is open for 5 months already, but there appears to be no progress yet. Can you give some information on when CubeCell support will be added?

ivankravets, 2020-03-05:
@wasn-eu @Heltec-Aaron-Lee could you help with integrating your software & hardware products?

Heltec-Aaron-Lee, 2020-03-06:
Is there any guy who can share some platformIO migration quick start manual? We will finish in the next two weeks.

lnlp, 2020-03-06:
Have a look here: https://docs.platformio.org/en/latest/platforms/custom_platform_and_board.html

ivankravets, 2020-03-07:
@Heltec-Aaron-Lee glad to see you here! Quick guide:

1) Create a new repository `https://github.com/HelTecAutomation/platform-asrmicro` in your organization and invite me (thanks!)
2) See examples of open source dev-platforms => https://github.com/topics/platformio-platform
3) Clone `https://github.com/HelTecAutomation/platform-asrmicro` to `~/.platformio/platforms/asrmicro`. You can work on integration here and commit changes to the repository.

This repository is a good starting point https://github.com/platformio/platform-atmelmegaavr
You just need to correct packages in `platform.json` to use ARM toolchain. Check other dev-platforms where we use this toolchain. For example, https://github.com/platformio/platform-atmelsam

Thanks!

eni23, 2020-04-24:
@Heltec-Aaron-Lee any update on this?

jacobsonjn, 2020-05-15:
No new updates on this?

samy4sam, 2020-06-05:
TThe ASR Microelectronics ASR605x platform and some CubeCell boards are available on Platformio since yesterday. But I get an error message when I try to install the ASR605x platform

`PIO Core Call Error: "PlatformManager: Installing asrmicro650x\r\n\n\nError: Detected unknown package 'asrmicro650x'"`

[https://docs.platformio.org/en/latest/platforms/asrmicro650x.html](url)

ivankravets, 2020-06-05:
We have not released it yet. We are in the process. Could you try the latest development version?

1. Please open PlatformIO IDE Terminal and type:
```
pio platform install https://github.com/HelTecAutomation/platform-asrmicro650x.git
```
2. Use any of these examples https://github.com/HelTecAutomation/platform-asrmicro650x/tree/develop/examples

Does it work? If yes, we will do the first 1.0.0 release. Thanks!

ivankravets, 2020-06-05:
We've just published 0.1.0 version into the our registry https://github.com/HelTecAutomation/platform-asrmicro650x/releases/tag/v0.1.0

Please try again.

closed 2020-06-05 by ivankravets