platformio/platformio-core #3079: insufficient hardware documentatio

mikronauts, 2019-10-02:
It is impossible to create new capabilities and supporting libraries for them without fully documenting the hardware subsystems like the DMA, SPI, UART peripherals with bit level documentations.

The current libraries only expose the functionalities the kendryte developers consider important, and while those are an excellent start, they lack outside the box possibilities.

Note bitluni ntsc, pal, vga etc work with the ESP32, which does have register level documentation.

If the concern is NDA information for the AI and crypto hardware, require NDA's for those peripheral blocks.

ivankravets, 2019-10-02:
Sorry, what is your dev/platform and framework/SDK?

mikronauts, 2019-10-02:
Hi,

https://docs.platformio.org/en/latest/platforms/kendryte210.html

I was looking into platformio.

My apologies for my ranting, I was upset at Kendryte responses to other developers asking for register level hardware documentation and posted in the wrong forum. This is NOT platformio's issue.

closed 2019-10-02 by mikronauts