platformio/platformio-core #3089: Lib installer hangs if project is on NFS mount

augonis, 2019-10-08:
### Configuration

**Operating system**: Ubuntu 18.04
**PlatformIO Version**: 4.1.0b3
### Description of problem
If the project is within a mounted NFS share library install hangs and must be killed

#### Steps to Reproduce

1. Make new project within a mounted NFS share
2. Try to install a library e.g. `pio lib install "ESPAsyncTCP"`
3. ctrl+C when you are tired of waiting
4. Wait some more for the process to terminate

### Actual Results
```
Library Storage: [somewhere in NFS mmount]
Looking for ESPAsyncTCP library in registry
Found: https://platformio.org/lib/show/305/ESPAsyncTCP [stuck here til I give up]
^C [waiting here for another good minute]
Aborted!
```

### Expected Results
Library successfully installed

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino
```

ivankravets, 2020-02-14:
Could you reproduce this issue with the latest PIO Core 4.2?

augonis, 2020-02-15:
Same results with PIO Core 4.2.0
I've tried to pause execution with pydev debugger and no reaction => python interpreter itself blocking on some system call?

ivankravets, 2020-02-15:
Could you help to debug this issue? I don't have a similar environment. The PIO Core source code is located in 
```
python -c "import platformio; print(platformio)"
```

augonis, 2020-02-15:
I'm open to help, with some guidance.

augonis, 2020-02-15:
I hooked up debugger an this time got some results.
In `lockfile.py` line 64 `fcntl.flock(self._fp.fileno(), fcntl.LOCK_EX | fcntl.LOCK_NB)` always raises `OSError No locks available`· 

Now seeing that error I remember another python based project (can't recall which) terminating with that error under similar circumstances. Could be python bug.

ivankravets, 2020-02-15:
Could you try to replace in this file all
```
fcntl.flock(
```

by

```
fcntl.lockf(
```
?

Does it work now?

See https://stackoverflow.com/questions/22409780/flock-vs-lockf-on-linux

augonis, 2020-02-15:
Same result.
Reading [https://serverfault.com/questions/66919/file-locks-on-an-nfs](https://serverfault.com/questions/66919/file-locks-on-an-nfs) and [https://unix.stackexchange.com/questions/396886/local-file-locking-on-nfs-being-linux-kernel-dependent](https://unix.stackexchange.com/questions/396886/local-file-locking-on-nfs-being-linux-kernel-dependent) I gather locks should work on NFS, but depends on server and client setup. 
As workaround I remounted the share with `nolock` option with positive results.

ivankravets, 2020-02-15:
> As workaround I remounted the share with nolock option with positive results.

I think this is the right solution? If we drop the blocking process of package installer, it will lead to a different problem when 1> concurrent instances tries to install the same package or manager registry.

Can we close this issue?

augonis, 2020-02-15:
Agreed. Close it.

ivankravets, 2020-02-15:
Happy coding with PlatformIO!

closed 2020-03-02 by ivankravets

osctobe, 2024-01-20:
In case this still is a problem, there is an old tool `dotlockfile` that does NFS-safe locking. Adding `nolock` to mount options means that the problem mentioned above -- concurrent instances, running on different NFS clients --
will stomp on each other.