platformio/platformio-core #3097: Core: Keyboard/HID doesn't work with Arduino Micro

Blockguy24, 2019-10-09:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): 4.0.3

### Description of problem
Keyboard and HID functions do not work with Arduino Micro. This is a similar issue to #881
I have tried using -DUSB_HID and -DUSB_SERIAL_HID but it still doesn't work.

#### Steps to Reproduce

1. Upload code that uses Keyboard.h library to Arduino Micro
2. Observe that it doesn't work and the Arduino is recognized as being in bootloader mode

### Actual Results

Code using Keyboard or HID libraries don't work and the Arduino Micro isn't detected properly.

Images:
[image]
[image]

### Expected Results

The code works as intended and the Arduino Micro is recognized properly

Images when compiled using Arduino IDE:
[image]
[image]

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:micro]
platform = atmelavr
board = micro
framework = arduino
```

**Source file to reproduce issue:**
```c++
#include <Arduino.h>
#include <Keyboard.h>

void setup() {
  pinMode(6, INPUT_PULLUP);
  Keyboard.begin();
}

void loop() {
  if (!digitalRead(6))
  {
    Keyboard.print("a");
  }
}
```

### Additional info

Interestingly enough, the Arduino is still able to run code when in this fake bootloader state, just HID functionality doesn't work.

Blockguy24, 2019-10-10:
So, I figured out what is happening.

If I put -DUSB_PID=0x8037 in the build flags, it still doesn't work, so I ran
`platformio run --target upload -v`
to see the command line arguments to the compiler, and here is what I found:

`avr-g++ -o .pio\build\micro\FrameworkArduino\PluggableUSB.cpp.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega32u4 -DPLATFORMIO=40003 -DARDUINO_AVR_MICRO -DUSB_PID=0x8037 -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO=10805 -DUSB_VID=0x2341 -DUSB_PID=0x0037 "-DUSB_PRODUCT=\"Arduino Micro\"" -DUSB_MANUFACTURER=\"Arduino\" "-IC:\Users\Joshua Star\.platformio\packages\framework-arduinoavr\cores\arduino" "-IC:\Users\Joshua Star\.platformio\packages\framework-arduinoavr\variants\micro" "C:\Users\Joshua Star\.platformio\packages\framework-arduinoavr\cores\arduino\PluggableUSB.cpp"`

Note the duplicated `-DUSB_PID=<PID>`, and that my user-defined PID is having no effect because the one after it is overriding it.

Blockguy24, 2019-10-16:
I found a workaround by editing the `micro.json` file in the `.platformio\platforms\atmelavr\boards` folder.
Old file:
```json
{
  "build": {
    "core": "arduino",
    "extra_flags": "-DARDUINO_AVR_MICRO",
    "f_cpu": "16000000L",
    "hwids": [
      [
        "0x2341",
        "0x0037"
      ],
      [
        "0x2341",
        "0x8037"
      ],
      [
        "0x2A03",
        "0x0037"
      ],
      [
        "0x2A03",
        "0x8037"
      ]
    ],
    "mcu": "atmega32u4",
    "usb_product": "Arduino Micro",
    "variant": "micro"
  },
  "frameworks": [
    "arduino"
  ],
  "fuses": {
    "efuse": "0xcb",
    "hfuse": "0xd8",
    "lfuse": "0xff",
    "lock": "0x3F"
  },
  "name": "Arduino Micro",
  "upload": {
    "disable_flushing": true,
    "maximum_ram_size": 2560,
    "maximum_size": 28672,
    "protocol": "avr109",
    "require_upload_port": true,
    "speed": 57600,
    "use_1200bps_touch": true,
    "wait_for_upload_port": true
  },
  "url": "https://www.arduino.cc/en/Main/ArduinoBoardMicro",
  "vendor": "Arduino"
}
```
New file:
```json
{
  "build": {
    "core": "arduino",
    "extra_flags": "-DARDUINO_AVR_MICRO",
    "f_cpu": "16000000L",
    "hwids": [
      [
        "0x2341",
        "0x8037"
      ],
      [
        "0x2341",
        "0x0037"
      ],
      [
        "0x2A03",
        "0x8037"
      ],
      [
        "0x2A03",
        "0x0037"
      ]
    ],
    "mcu": "atmega32u4",
    "usb_product": "Arduino Micro",
    "variant": "micro"
  },
  "frameworks": [
    "arduino"
  ],
  "fuses": {
    "efuse": "0xcb",
    "hfuse": "0xd8",
    "lfuse": "0xff",
    "lock": "0x3F"
  },
  "name": "Arduino Micro",
  "upload": {
    "disable_flushing": true,
    "maximum_ram_size": 2560,
    "maximum_size": 28672,
    "protocol": "avr109",
    "require_upload_port": true,
    "speed": 57600,
    "use_1200bps_touch": true,
    "wait_for_upload_port": true
  },
  "url": "https://www.arduino.cc/en/Main/ArduinoBoardMicro",
  "vendor": "Arduino"
}
```

ivankravets, 2019-10-24:
Thanks! Fixed in https://github.com/platformio/platform-atmelavr/commit/0c0972b8e5216cc4e87b4292dbe4fb66d58c8a83

closed 2019-10-24 by ivankravets

akshaygugnani15, 2020-07-07:
You know what you don't require the keyboard.h or 32u4 now, I got a solution to use the keyboard and mouse through any Arduino yes you read it right any!

All you have to do is something like this on the Arduino:-

`Serial.println("pressA")`

And now head to a python script and do this:-

```
import serial
import pyautogui
Arduino_Serial = serial.Serial('COM5', 9600)

while 1:
    incoming_data = str(Arduino_Serial.readline())
    print(incoming_data)
    if 'pressA' in incoming_data:
        pyautogui.press('a')
    incoming_data = ""
```
BOOM YOU ARE DONE