platformio/platformio-core #3101: Home: Libraries: Could not update library

de-eb, 2019-10-10:
PIO Core Call Error: "Library Storage: C:\\Users\\mito-6\\OneDrive\\ドキュメント\\Research\\make\\Source\\Anemon\\Anemon_Sensor\\.pio\\libdeps\\m5stack-core-esp32\n\n\nc:\\users\\mito-6\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py:425: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal\n  if manifest['__pkg_dir'] == abspath(pkg_dir):\nError: Traceback (most recent call last):\n  File \"c:\\users\\mito-6\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 90, in main\n    cli(None, None, None)\n  File \"c:\\users\\mito-6\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 764, in __call__\n    return self.main(*args, **kwargs)\n  File \"c:\\users\\mito-6\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 717, in main\n    rv = self.invoke(ctx)\n  File \"c:\\users\\mito-6\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 37, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"c:\\users\\mito-6\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\mito-6\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\mito-6\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 956, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"c:\\users\\mito-6\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 555, in invoke\n    return callback(*args, **kwargs)\n  File \"c:\\users\\mito-6\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 17, in new_func\n    return f(get_current_context(), *args, **kwargs)\n  File \"c:\\users\\mito-6\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib.py\", line 253, in lib_update\n    lm.update(library, only_check=only_check)\n  File \"c:\\users\\mito-6\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 773, in update\n    pkg_dir = self.get_package_dir(*self.parse_pkg_uri(package))\n  File \"c:\\users\\mito-6\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 237, in parse_pkg_uri\n    text = str(text)\nUnicodeEncodeError: 'ascii' codec can't encode characters in position 25-30: ordinal not in range(128)\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

ivankravets, 2019-10-24:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
python --version
pip --version
pip list
```

de-eb, 2019-10-26:
C:\WINDOWS\system32>echo %PATH%
C:\Program Files\Python37\Scripts\;C:\Program Files\Python37\;C:\Program Files (x86)\Python37-32\Scripts\;C:\Program Files (x86)\Python37-32\;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\WINDOWS\System32\OpenSSH\;C:\texlive\2018\bin\win32;C:\Program Files\Git\cmd;C:\tools\OpenOCD\0.10.0-12-20190422-2015\bin;C:\Users\Takuya\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files (x86)\GnuWin32\bin;C:\tools\gcc-arm-none-eabi-8-2018-q4-major-win32\bin;C:\MinGW\bin

C:\WINDOWS\system32>python --version
Python 3.7.4

C:\WINDOWS\system32>pip --version
pip 19.2.3 from c:\program files\python37\lib\site-packages\pip (python 3.7)

C:\WINDOWS\system32>pip list
Package           Version
----------------- -------------
absl-py           0.8.0
altgraph          0.16.1
astor             0.8.0
astroid           2.3.1
boto3             1.9.243
botocore          1.12.243
certifi           2019.9.11
chardet           3.0.4
colorama          0.4.1
configparser      4.0.2
contextlib2       0.6.0
cycler            0.10.0
Cython            0.29.13
docutils          0.15.2
future            0.17.1
gast              0.3.2
grpcio            1.24.1
h5py              2.10.0
hidapi            0.7.99.post21
idna              2.8
imageio           2.5.0
isort             4.3.21
jmespath          0.9.4
joblib            0.14.0
kiwisolver        1.1.0
lazy-object-proxy 1.4.2
macholib          1.11
Markdown          3.1.1
matplotlib        3.1.1
mccabe            0.6.1
mock              3.0.5
numpy             1.17.2
opencv-python     4.0.0.21
packaging         19.2
pandas            0.24.2
pefile            2019.4.18
pip               19.2.3
pip-review        1.0
protobuf          3.8.0
pygame            1.9.6
PyInstaller       3.4
pylint            2.3.1
pyparsing         2.4.2
PyQt5             5.12.2
PyQt5-sip         4.19.17
pyserial          3.4
python-dateutil   2.8.0
pytz              2019.1
pywin32-ctypes    0.2.0
PyYAML            5.1.2
requests          2.22.0
rope              0.14.0
s3transfer        0.2.1
scikit-learn      0.21.2
scipy             1.3.1
setuptools        41.2.0
six               1.12.0
sklearn           0.0
termcolor         1.1.0
tqdm              4.36.1
typed-ast         1.4.0
urllib3           1.25.6
Werkzeug          0.15.4
wheel             0.33.4
wrapt             1.11.2
xlrd              1.2.0

Thank you for your response! : )

ivankravets, 2019-10-26:
Does you project path contain non-ascii chars?

ivankravets, 2019-10-31:
We did some improvements on PIO Core side. Please re-test with `pio upgrade --dev`.

closed 2019-10-31 by ivankravets