platformio/platformio-core #3113: Replace LDF deep with native GCC Preprocessor

ivankravets, 2019-10-13:
Subj.

italocjs, 2023-09-03:
Hello, i have seen multiple issues (mentioned here), and the only workaround that worked for me was adding lib_ldf_mode = deep

Is this the best approach for an workaround?  Is there any planning to work on this issue?

ivankravets, 2023-09-04:
@italocjs, yes, this is a good temporary solution. Another way is to directly declare dependencies in `lib_deps`. See https://docs.platformio.org/en/latest/projectconf/sections/env/options/library/lib_deps.html

brentru, 2023-09-26:
@ivankravets Is there any current PRs/work on this issue or is it no longer a priority since explicit declaration into lib_deps and using a different LDF mode is a valid workaround.

JustTryingToGetSomeWorkDone, 2023-12-01:
I'm still experiencing this issue and it prevents me from using VS Code\PlatformIO for embedded programming. If I can't compile, it's useless. I'm not going to spend hours setting up every library with workarounds for every little project that comes my way. My time spent working around my libraries not working will quickly outpace my time spent programming.

ivankravets, 2023-12-01:
@JustTryingToGetSomeWorkDone , could you provide more details on your issue?

navado, 2024-01-29:
@ivankravets I'm experiencing same issue.
Please find details below:
# The error:
```
In file included from lib/Adafruit Sensor Calibration/Adafruit_Sensor_Calibration.h:61:0,
                 from lib/Adafruit Sensor Calibration/Adafruit_Sensor_Calibration.cpp:1:
lib/Adafruit Sensor Calibration/Adafruit_Sensor_Calibration_SDFat.h:14:10: fatal error: Adafruit_SPIFlash.h: No such file or directory

***************************************************************************
* Looking for Adafruit_SPIFlash.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_SPIFlash.h"
* Web  > https://registry.platformio.org/search?q=header:Adafruit_SPIFlash.h
*
***************************************************************************

 #include <Adafruit_SPIFlash.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

# PlatformIO INI:
```

[env]
platform = nordicnrf52
board = adafruit_cplaynrf52840
framework = arduino
lib_ignore = Adafruit BluefruitLE nRF51
lib_ldf_mode = chain+
lib_deps = 
	adafruit/Adafruit LED Backpack Library@1.5.0
	adafruit/Adafruit Circuit Playground@^1.12.0
	adafruit/Adafruit Unified Sensor@^1.1.14
    adafruit/Adafruit Sensor C
    adafruit/SdFat - Adafruit Fork@^2.2.3
    adafruit/Adafruit NeoPixel@^1.12.0
    adafruit/Adafruit AHRS@^2.3.5
    adafruit/Adafruit BusIO@^1.15.0
    adafruit/Adafruit GFX Library@^1.11.9
    adafruit/Adafruit SPIFlash@^4.3.4

[env:empty]
build_flags = -D ENV_EMPTY
build_src_filter = +<*> -<examples> -<examples/*> -<test> -<test/*> -<tests> -<tests/*> -<test_*> -<main-*> +<main-empty.cpp>

```

# main-empty.cpp

```C++
#include <Adafruit_CircuitPlayground.h>

void setup() {
  CircuitPlayground.begin();
}

void loop() {
  CircuitPlayground.redLED(HIGH);
  delay(123);
  CircuitPlayground.redLED(LOW);
  delay(345);
}
```

ScaredyCat, 2024-05-31:
Running into this issue. compiling header files that aren't used and have other dependancies means I have to download the lib, put it in my project libs folder and delete the files I'm not using and aren't required.

Example: including TLog library ( https://registry.platformio.org/libraries/makerspaceleiden/TLog ):

`#include <TLog.h>

#include <TelnetSerialStream.h>
TelnetSerialStream telnetSerialStream = TelnetSerialStream();`

When building, WebSerialStream.h /cpp are included, though not needed for this build, adding another library dependancy for WebServer.h which is also not required. 

`In file included from .pio/libdeps/client/TLog/src/WebSerialStream.cpp:21:
.pio/libdeps/client/TLog/src/WebSerialStream.h:28:10: fatal error: WebServer.h: No such file or directory`

Makes the built in library management a lot less useful.