platformio/platformio-core #3116: Home: Could not initialize project

BriamCruz, 2019-10-14:
PIO Core Call Error: "The current working directory C:\\Users\\Briam\\Documents\\HI will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\Briam\\Documents\\HI\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Detected unknown package 'tool-scons'"

trylika, 2019-10-14:
duplicate for #3115 #3110 #3108 #3094 #3090? please check other issues before posting

ivankravets, 2019-10-24:
Please reopen if you still need help

closed 2019-10-24 by ivankravets

bridystone, 2020-01-14:
i'm running OS X 10.10 and i've installed newly VSCode & platformio.

if i want to create a new project, i'm told that platform cannot install the atmelavr package (or any other package - i've tested several).

installing the package via CLI doesn't work either:
PlatformManager: Installing atmelmegaavr
Error: Detected unknown package 'atmelmegaavr'

What is causing the problem?

ivankravets, 2020-01-14:
Please install the latest Python from official web site. Remove `~/.platformio/penv` folder and restart VSCode. 

Please note that you need also to install SSL certificates. The installation script should be in mounted Python image.

bridystone, 2020-01-15:
hi @ivankravets,

Thanks for the fast reply.

I've installed the most recent version of python from python.org.
(it's a pkg file without any mounted image) 
I've also installed pip with this command:
"sudo curl https://bootstrap.pypa.io/get-pip.py | python3".

Then i deleted the mentioned folder and restarted VSCode (the part of installation was handled again)

however it didn't change the behavior. 
Any other suggestion?
Any log file i could send?
---
Python 3.8.1
---
pip 19.3.1 from /Library/Python/2.7/site-packages/pip (python 2.7) 
pip 19.3.1 from /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pip (python 3.8)
---

BR
Jens

ivankravets, 2020-01-15:
https://www.python.org/downloads/release/python-381/

```
macOS users

    For Python 3.8.0, we provide one installer: 64-bit-only that works on macOS 10.9 (Mavericks) and later systems.
    Please read the "Important Information" displayed during installation for information about SSL/TLS certificate validation and the running the "Install Certificates.command".
```

bridystone, 2020-01-15:
Wow! Thanks that helped!!

I shouldn't have just clicked away the "Important Information".
Usually this is only some nonsense...

Again - Thanks very much for your fast support!

Jens

ivankravets, 2020-01-15:
We will think about how to help to avoid this mistake. We actually can't run this script because it requires super user's permissions

Happy coding with PlatformIO.

bridystone, 2020-01-15:
The Error message could hint in this direction.

Then at least the way forward would be clear.
The general error message "unknown package detected" is a little misleading... :)

BTW: coding with platformio rocks - for the couple of hours i now had :)

AGKAGK, 2020-01-27:
Hello, I get almost the same warning *Could not initialize the project". First small problem, I cannot copy this (and paste this long message).  The Massage is ending with ... Detected unknown package 'atmelavr".  My MBP running with 10.11.6.
I press button "Check available solutions"  find this.  Python 3.8.1 is installed.  The "Install Certificates.command" was running.  
I don't know how to use the command line solutions above, on left side I find some Terminals... (?).
I have tried some things, but I don't know if it is right. What is meant by "...Misc and to install required dev/platform manually: ".
I cannot delete ~/.platformio/penv because I cannot find it (hidden files are visible).

Andreas

bridystone, 2020-01-27:
Hi AGKAGK,

you can enter the folder by "goto" in your finder menu and then "goto Folder"

Navigate your Finder to your Home-Folder.
Open the Menu "Goto" and choose "Goto Folder" (2nd to last item)
here you type ".platformio" - enter

now you are in the hidden folder....
Hope this helps

Jens

AGKAGK, 2020-01-28:
Halo Jens, l

prima, dank, darüber habe ich es gefunden. (Das nutze ich sonst nicht, ich hatte die Suche verwendet.)
Ich bin (vermutlich) auch einen Schritt weiter, aber gleich gestern Abend, und so auch heute früh bekam ich diese Meldung:

[Error - 8:03:17 AM] Connection to server got closed. Server will not be restarted.

Liegt das nun am Server oder meinem System?

Is the reason for the message the server or my system?

Andreas

AGKAGK, 2020-01-28:
P.S. Firewalls are off  (Router+MBP)

ivankravets, 2020-01-28:
@AGKAGK could you run this?

```
python3 -c "from urllib.request import urlopen; print(urlopen('https://github.com').getcode())"
```

Do yo usee `200`?

AGKAGK, 2020-01-28:
Thanks ivankravets,

please tell me where to enter this code? And which value you mean with "200"?

ivankravets, 2020-01-28:
In system terminal

AGKAGK, 2020-01-28:
It is 200.

AGKAGK, 2020-02-05:
Hello, its the same. How ist the stand?

AndreVanKammen, 2022-01-05:
Hello,

just wanted to share this experience, I will update this post with the answer for future reference since my search ended here.

I'm runnig on windows 10.  was working happily in an old environment of Platform.io to update some old sketch. I could build and upload my project, VS-Code mentioned to install the Platform-IO extention so I did that. It started to update everything including the commandline version. And now i get this on "pio run"

```
************************************************************************************************************************************************
Obsolete PIO Core v3.5.3 is used (previous was 5.2.4)
Please remove multiple PIO Cores from a system:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
************************************************************************************************************************************************ 
[01/05/22 20:05:27] Processing esp32dev (framework: arduino; platform: https://github.com/platformio/platform-espressif32.git#feature/stage; board: esp32dev)
------------------------------------------------------------------------------------------------------------------------------------------------ 
CorePackageManager: Installing tool-scons @ ~2.20501.4
Error: Detected unknown package 'tool-scons'
```

The link is outdated it needs to be
  https://docs.platformio.org/en/latest//faq.html#multiple-platformio-cores-in-a-system
  
But i guess it's from the old version I use.

averter, 2023-09-06:
To summarise and if you are on linux, run either `pip uninstall platformio` or `sudo apt-get purge platformio`. Then run `pip3 install platformio`. Afterwards do `pio platform install atmelavr`

ivankravets, 2023-09-07:
`pio platform` is deprecated and will be removed soon. See https://docs.platformio.org/en/latest/core/userguide/pkg/index.html