platformio/platformio-core #3120: Home: Could not load PIO Account information

RobGannon, 2019-10-14:
PIO Core Call Error: "the JSON object must be str, not 'bytes'"

ivankravets, 2019-10-24:
What is your OS?

takanotume24, 2019-10-25:
I have same problem.
```
takuya@robobu-ubuntu-2600k:~$ uname -a
Linux robobu-ubuntu-2600k 4.4.0-166-generic #195-Ubuntu SMP Tue Oct 1 09:35:25 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

ivankravets, 2019-10-28:
1. Please open system task manager and kill all python.exe processes
2. Run `pio update` in PlatformIO IDe Terminal.

Does it work now?

ivankravets, 2019-10-29:
Please reopen if you still need help

closed 2019-10-29 by ivankravets

burnsed, 2019-12-13:
Hi,
I'm also seeing this issue. I tried updating via `pio update` within the vscode terminal and then I tried following the instructions in  issue #3219 of upgrading to dev `pio upgrade --dev`. I still experience the same problem after both. The pio and python versions are:

```
$ ~/.platformio/penv/bin/python --version
Python 3.5.3
$ ~/.platformio/penv/bin/pio --version
PlatformIO, version 4.1.1b5
```

When running `pio home` from vscode, this is the error I get in the terminal:
`Error: {'data': "the JSON object must be str, not 'bytes'", 'code': 4003, 'message': 'PIO Core Call Error'}`

Additionally, in case it is related, I'm using my account so that I can upload via a remote agent (which is also upgraded to the same version), when I try to list devices I also get a similar error:
```
$ pio remote device list
PIO Plus (https://pioplus.com) v2.6.1
sequence item 2: expected str instance, bytes found
```
 
Thanks

ivankravets, 2019-12-21:
Could you run from your CLI:
```
pio account show --json-output
```

Do you see any errors?

IlnurKashapov, 2020-02-26:
Hi.
I'm also seeing this issue. 
```
~/.platformio/penv/bin/python --version
Python 3.5.2
~/.platformio/penv/bin/pio --version
PlatformIO, version 4.2.2a1
```
```
~ $  uname -a
Linux ilnur-comp 4.4.0-21-generic #37-Ubuntu SMP Mon Apr 18 18:33:37 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

```

When I run `pio account show --json-output` within the vscode terminal:

`{"username": "****@***.***", "subscriptions": [], "currentPlan": "PIO Plus Community", "groups": [{"permissions": ["OTA Device Manager", "Remote Agent List", "100 Shares per Remote Agent", "100 Concurrent Remote Agents", "Remote Serial Port Monitor (50,000 per month)", "Remote Firmware Updates (100,000 per month)", "Remote Unit Testing (100,000 per month)"], "name": "PIO Plus Community", "expire": 0}], "upgradePlan": "PIO Plus Professional for Individual"}`

ivankravets, 2020-02-29:
@IlnurKashapov could you reproduce this issue after VSCode restart?

IlnurKashapov, 2020-03-03:
> @IlnurKashapov could you reproduce this issue after VSCode restart?

It works now. thanks!