platformio/platformio-core #3123: Home: Libraries: Could not install library

miteshdube, 2019-10-15:
PIO Core Call Error: "Library Storage: /home/mitesh/.platformio/lib\nLibraryManager: Installing id=5437 @ ^1.3.0\n\n\nError: Traceback (most recent call last):\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/platformio/__main__.py\", line 90, in main\n    cli(None, None, None)\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/click/core.py\", line 764, in __call__\n    return self.main(*args, **kwargs)\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/click/core.py\", line 717, in main\n    rv = self.invoke(ctx)\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/platformio/commands/__init__.py\", line 37, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/click/core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/click/core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/click/core.py\", line 956, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/click/core.py\", line 555, in invoke\n    return callback(*args, **kwargs)\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/click/decorators.py\", line 17, in new_func\n    return f(get_current_context(), *args, **kwargs)\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/platformio/commands/lib.py\", line 150, in lib_install\n    force=force)\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/platformio/managers/lib.py\", line 359, in install\n    force=force)\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/platformio/managers/package.py\", line 703, in install\n    pkg_dir = self._install_from_piorepo(name, requirements)\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/platformio/managers/lib.py\", line 220, in _install_from_piorepo\n    if app.get_setting(\"strict_ssl\") else dl_data['url'], requirements)\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/platformio/managers/package.py\", line 481, in _install_from_url\n    dlpath = self.download(url, tmp_dir, sha1)\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/platformio/managers/package.py\", line 179, in download\n    fd.start(with_progress=with_progress)\n  File \"/home/mitesh/.platformio/penv/lib/python3.5/site-packages/platformio/downloader.py\", line 83, in start\n    f.write(next(itercontent))\nStopIteration\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

miteshdube, 2019-10-15:
Any solution for this problem.

ivankravets, 2019-10-17:
Duplicate of https://github.com/platformio/platformio-core/issues/2717

closed 2019-10-17 by ivankravets