platformio/platformio-core #3128: Home: Could not load boards list

nicolasdelta, 2019-10-15:
PIO Core Call Error: "Error: Traceback (most recent call last):\n  File \"c:\\users\\nicolas\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 90, in main\n    cli(None, None, None)\n  File \"c:\\users\\nicolas\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 764, in __call__\n    return self.main(*args, **kwargs)\n  File \"c:\\users\\nicolas\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 717, in main\n    rv = self.invoke(ctx)\n  File \"c:\\users\\nicolas\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 37, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"c:\\users\\nicolas\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\users\\nicolas\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 956, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"c:\\users\\nicolas\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 555, in invoke\n    return callback(*args, **kwargs)\n  File \"c:\\users\\nicolas\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 31, in cli\n    return _print_boards_json(query, installed)\n  Fil

Vik137, 2019-10-16:
> Executing task in folder 191016-181747-uno: C:\Users\User\.platformio\penv\Scripts\platformio.exe debug <

Error: Currently, PlatformIO does not support debugging for `Arduino Uno`.     
Please request support at https://github.com/platformio/platformio-core/issues 
or visit -> https://docs.platformio.org/page/plus/debugging.html
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

ivankravets, 2019-10-24:
Yes, there is no debug for Uno yet.

closed 2019-10-24 by ivankravets