platformio/platformio-core #3132: Request for support to pass extra arguments to the target

KKoovalsky, 2019-10-16:
- [ ] **Feature Request**.

I would like to be able to pass extra arguments to a binary e.g. when running on a native platform.

Possible usage, when running [Catch2](https://github.com/catchorg/Catch2) tests:

```
platformio test -e native --extra-args="[testA][testB]"
```

nbolton, 2022-04-29:
From https://github.com/platformio/platformio-core/issues/4248:
> Within my project, I have many test functions. When one fails, finding out which one is tedious. Unity has command line support (when UNITY_USE_COMMAND_LINE_ARGS is defined) through it's -n or -f arg, but I couldn't see how to pass args to Unity via pio. The test_filter option in the .ini isn't useful in this situation, because it filters on test library, not test function.
> 
> Idea: Provide a way of filtering by Unity test function, or maybe have a way of passing args to Unity.

ivankravets, 2022-05-07:
Resolved in https://github.com/platformio/platformio-core/commit/daa34818622f09b2c9fa7743289f3d54bc46edf0

The new `-a, --program-arg` option has been added to the `pio test` command. Please re-test. See also updated docs.

closed 2022-05-07 by ivankravets

robsonos, 2023-11-02:
Hi all

I know this is an old ticked, but I think my comments are relevant here.
I need to parse args from `pio test`  in a non-native environment, and it looks like the solution here provided that. However, in the latest version of PlatromIO , calling `--program-arg` with `pio test`  does not set `PROGRAM_ARGS`. 
Did you guys create an alternative for that that I am missing or my understanding is correct?

Cheers