platformio/platformio-core #3140: Home: Could not load boards list

Hernan2882, 2019-10-17:
PIO Core Call Error: "cannot import name threadpool"

Megatrons-cannon, 2019-10-19:
Same issue here as well

ivankravets, 2019-10-20:
What is your OS?

ivankravets, 2019-10-23:
Please reopen if you still need help

closed 2019-10-23 by ivankravets

redquartista, 2019-11-03:
Getting same issue on trying to create new project. Unresolved even after updating PIO. Any leads?

ivankravets, 2019-11-03:
Could you reproduce this issue after VSCode restarting?

redquartista, 2019-11-03:
Yes. It pops up every time I try to create new project.  It looks quite like this: 

[image]

ivankravets, 2019-11-05:
@redquartista could you provide AnyDesk remote session to debug this issue? You can send session id to me@ikravets.com. Thanks!

redquartista, 2019-11-05:
> @redquartista could you provide AnyDesk remote session to debug this issue? You can send session id to [me@ikravets.com](mailto:me@ikravets.com). Thanks!

Hi, today, it did not display that error. I have been able to create new project and browse all the boards. I did not do anything significant. The computer was manually restarted in a totally unrelated step to this PIO/VSCode. And it worked. :)